project-chip/connectedhomeip #21744: about leave event

Suguotao, 2022-08-09:
No matter how many fabrics the dut joins,dut just report fabricIndex 1 when i call chip::Server::GetInstance().GetFabricTable().DeleteAllFabrics() before factroy reset
  *expected behavior
  report all fabric index
 * actual behavior
  report fabric index1
* steps to reproduce
  first : dut commission to TH1 and TH2
  second : TH1 and TH2 subscribe leave event: basic subscribe-event leave 1 60 1 0、basic subscribe-event leave 1 60 2 0 --commissioner-name beta
  third : call chip::Server::GetInstance().GetFabricTable().DeleteAllFabrics() befofe factory reset

I try to understand the code and I can see that the event has been pushed to eventManagerment,but it doesn't work well actually, I don't understand what's going on here

bzbarsky-apple, 2022-08-09:
What I would expect to happen here is:

1. Fabric index 1 gets removed.  This queues a Leave event, which is reported to both subscriptions.
2. Fabric index 2 gets removed.  This queues a second Leave event, which is reported to only fabric 2 (because fabric 1 is gone by this point).

I am seeing item 1 happen, but not seeing the second Leave event for item 2.  Looking into why not.

bzbarsky-apple, 2022-08-09:
Ah, I think the issue is that the ReadHandler for the fabric 2 subscription has sent out the Leave event for fabric 1 and has not gotten the Status Response yet, so it's not able to send any more data reports.

It seems like we should only be trying to sync-deliver Leave _on the fabric being left_.  So `Engine::ScheduleUrgentEventDeliverySync` needs to grow a way to only do sync delivery on a specific fabric index, perhaps....

closed 2022-08-10 by woody-apple