project-chip/connectedhomeip #21746: [TC-DRLK-2.1] When we try to unlock the door, lock state is not getting changed

temedi, 2022-08-09:
**Description:** When we try to unlock the door, lock state is not getting changed

**Steps to reproduce the issue:**
**Step 1.** ./chip-tool doorlock read lock-state 1 1

**Step 2.**  ./chip-tool doorlock  unlock-door 1 1  --timedInteractionTimeoutMs 100 

**Step 3.** ./chip-tool doorlock read lock-state 1 1

Expected outcome: Lock State value should be 2. 

Actual Outcome: Lock state is shown as 1

Logs:
[2_1_1e_chiptool_log.txt](https://github.com/project-chip/connectedhomeip/files/9289054/2_1_1e_chiptool_log.txt)
[2.1_1e_Device_log.txt](https://github.com/project-chip/connectedhomeip/files/9289076/2.1_1e_Device_log.txt)

Note: This issue is tested on https://github.com/project-chip/connectedhomeip.git  repo
**Commit ID used :**  83fb7fd78e10841b4ae7a4e9b0603fc3e1ce8c72

Thanks and Regards,
Tejaswi

bzbarsky-apple, 2022-08-09:
@temedi Which exact server app are you testing?  The one in `examples/lock-app/efr32`?

If that's the one, then its `setLockState` never actually sets the state....  @mykrupp

mykrupp, 2022-08-09:
@temedi @bzbarsky-apple There is a PR that fixed this for efr32 that went in today.

temedi, 2022-08-10:
> @temedi Which exact server app are you testing? The one in `examples/lock-app/efr32`?
> 
Yes we are using `examples/lock-app/efr32` for testing.

temedi, 2022-10-12:
When unlocking the door, lock state is getting updated to 2, Hence closing the Issue.

closed 2022-10-12 by temedi