project-chip/connectedhomeip #21763: Linksys MR9600 has issues commissioning a Wi-Fi/BT DUT to the test harness

ryetree1, 2022-08-09:
Have a Wi-Fi/BT DUT that has no issues commissioning to the TH with other routers and home networks.  One internal employee has a Linksys MR9600 with stock settings that cant seem to get it commissioned.  DUT connects to Wi-Fi, gets a IPv4 and IPv6 address, but the TH fails at the FindOperational step.

DUT Logs (application specific logs):
Interface 0 IP address : 10.0.0.77
Wi-Fi Link Status: CONNECTED
Interface 0 IPv6 address : 2600:8807:4881:a40f:0207:a6ff:fe2d:a66

Partial RPI Logs:
  IPv4 address for eth0:            10.0.0.215 
  IPv6 address for eth0:            2600:8807:4881:a40f:e65f:1ff:fe3f:ab18 

Received Command Response Data, Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0007 
[1660006080.828860][2321:2326] CHIP:CTL: Received ConnectNetwork response, networkingStatus=0 
[1660006080.828930][2321:2326] CHIP:CTL: Successfully finished commissioning step 'WiFiNetworkEnable' 
[1660006080.828990][2321:2326] CHIP:CTL: Commissioning stage next step: 'WiFiNetworkEnable' -> 'FindOperational' 
[1660006080.829056][2321:2326] CHIP:CTL: Performing next commissioning step 'FindOperational' 
[1660006080.829104][2321:2326] CHIP:CSM: FindOrEstablishSession: PeerId = [1:0000000000BC5C01] 
[1660006080.829159][2321:2326] CHIP:CSM: FindOrEstablishSession: No existing OperationalDeviceProxy instance found 
[1660006080.829269][2321:2326] CHIP:CTL: OperationalDeviceProxy[1:0000000000BC5C01]: State change 1 --> 2 
[1660006080.829342][2321:2326] CHIP:DIS: Resolving 6DDFE03B9F3EB60A:0000000000BC5C01 ... 
[1660006080.831850][2321:2326] CHIP:DIS: OperationalDeviceProxy[1:0000000000BC5C01]: Operational node lookup already in progress. Will NOT start a new one. 
[1660006080.832010][2321:2326] CHIP:DMG: ICR moving to [AwaitingDe] 
[1660006081.029607][2321:2326] CHIP:DIS: Checking node lookup status after 200 ms 
[1660006083.408121][2321:2326] CHIP:DL: HandlePlatformSpecificBLEEvent 16387 
[1660006085.834071][2321:2326] CHIP:DL: Re-trying resolve 
[1660006085.894406][2321:2324] CHIP:DL: Indication received, conn = 0xffff7c03f290 
[1660006085.894680][2321:2326] CHIP:DL: HandlePlatformSpecificBLEEvent 16389 
[1660006088.477304][2321:2326] CHIP:DL: HandlePlatformSpecificBLEEvent 16387 
[1660006090.838037][2321:2326] CHIP:DL: Re-trying resolve 
[1660006090.964529][2321:2324] CHIP:DL: Indication received, conn = 0xffff7c03f290 
[1660006090.964817][2321:2326] CHIP:DL: HandlePlatformSpecificBLEEvent 16389 
[1660006093.548151][2321:2326] CHIP:DL: HandlePlatformSpecificBLEEvent 16387 
[1660006095.831642][2321:2326] CHIP:DIS: Checking node lookup status after 15002 ms 
[1660006095.831774][2321:2326] CHIP:DIS: OperationalDeviceProxy[1:0000000000BC5C01]: operational discovery failed: ../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:174: CHIP Error 0x00000032: Timeout

Test harness
RPI4 - 8GB
Tried TE9, SVE1.4, and v2.1 versions.

s-mcclain, 2022-08-10:
If you have a tcpdump of the devices on your WiFi network, do you happen to see if the RPi received any mDNS responses to it's query during operational discovery?

tcarmelveilleux, 2022-08-12:
This is likely an mDNS issue on IPv6 or an issue with the DUT not properly using MLD protocol to join multicast groups in its IPv6 middleware, and the access point only forwarding IPv6 multicast traffic to nodes that have joined MLD group.

To diagnose more, we need to get a packet capture from the raspi at the very least, but the correct capture would be a link-layer packet capture on the Wi-Fi network, pre-decrypted. 

Assiging to @ryetree1 to get logs.

Overall, **this is a Wi-Fi interop issue between your DUT and that router** and not an issue with Matter stack or Matter protocol, until proven otherwise. It's not Matter's fault if a Wi-Fi access point or a particular product are not properly using IPv6 technology.

ryetree1, 2022-08-12:
I am working on acquiring the logs, and inquiring about the MLD protocol to our platform vendor (their wifi stack).   The logs will take a few more days, as this environment is not mine, nor is near me.

ryetree1, 2022-08-16:
James Harrow suggested on slack the issue may be the router is failing to bridge the ethernet and wifi networks.  The RPI is on ethernet, and DUT on wifi.   I could find no settings in the router to change how this is bridged.  Instead I joining the RPI via wifi, and the problem went away.  The DUT can now consistently join the matter RPI TH.

We put the RPI back on ethernet and did a wifi sniffing of the DUT joining wifi, and subsequent traffic.  See attached capture.
[MR9600_Matter_Join_Capture.zip](https://github.com/project-chip/connectedhomeip/files/9352277/MR9600_Matter_Join_Capture.zip)

It does not seem like this issue involves the Matter SDK, but rather a router that may struggle with this use case.  I suspect this will be one of many routers that may have this issue.  Some may be able to be fixed with router settings, if those settings are exposed.

closed 2023-03-12 by stale[bot]