project-chip/connectedhomeip #21783: Socket service is not connected if keep app backgroup for a while

JourneyYoung, 2022-08-10:
Hello. When we test our app with chip-tool on darwin, if we keep app on backgroud for a while then make it to the front, the socket service can never connected again, and matter framework will loop log print "Socket is not connected", see as the picture. This issue will make us can not pair any matter device unless we restart the app. 
So I want to know how to fix this?

[image]

JourneyYoung, 2022-08-10:
We use iPhone12 running on iOS 15.4

bzbarsky-apple, 2022-08-10:
This is a known limitation of the current iOS CHIPTool.  See also https://github.com/project-chip/connectedhomeip/issues/584

JourneyYoung, 2022-08-10:
您好。你的邮件我已经收到，谢谢！

bzbarsky-apple, 2022-08-10:
And right now the fix is "restart the app".

JourneyYoung, 2022-08-14:
Now this issue will call iOS App OOM crash,  how can I close socket service when the app on backgroup?

woody-apple, 2022-09-06:
Darwin Issue Review: Not critical for 1.0, moving to 1.1

woody-apple, 2023-04-04:
Dup of: https://github.com/project-chip/connectedhomeip/issues/584

closed 2023-04-04 by woody-apple