project-chip/connectedhomeip #21791: Wrong error code set in GetWeekDaySchedule for missing user

bzbarsky-apple, 2022-08-10:
#### Problem
Door lock spec, in "5.2.4.16.1. Status" says:

> NOT_FOUND if both WeekDayIndex and UserIndex are within the valid range, however, there is not corresponding schedule entry found.

but what the SDK implements in `DoorLockServer::getWeekDayScheduleCommandHandler` has:

```
    if (!userExists(endpointId, userIndex))
    {
        emberAfDoorLockClusterPrintln("[GetWeekDaySchedule] User does not exist [endpointId=%d,weekDayIndex=%d,userIndex=%d]",
                                      endpointId, weekDayIndex, userIndex);
        sendGetWeekDayScheduleResponse(commandObj, commandPath, weekDayIndex, userIndex, DlStatus::kFailure);
        return;
    }
```

which is using FAILURE instead of NOT_FOUND, afaict.

It's not clear to me whether there is a certification test for this...

@mykrupp @cjandhyala @naveenkommareddi 

 #### Proposed Solution
Double-check that I am not missing something, and if not fix the error code.

mykrupp, 2022-08-10:
I think the spec is not clear as to what happens if the user index is in a valid range, but there is no user in the index. The spec doesn't say it, but shouldn't this return NOT_FOUND? Or maybe USER_NOT_FOUND?

The other potential issue is that checking whether or not the schedule at the current given index is done in the Linux example, when it should be done in door-lock-server unless I am missing something. (i.e. that it was deliberately done this way). But seems unlikely as door-lock-server has a userExists function. Seems that the solution should be adding a scheduleExists function to door-lock-server.

mykrupp, 2022-08-10:
More succinctly - What happens when a user index is in valid bounds, but there is no user found there? What is the response? Is this defined in the spec?

bzbarsky-apple, 2022-08-10:
Again, the spec says:

>   NOT_FOUND if both WeekDayIndex and UserIndex are within the valid range, however, there is not corresponding schedule entry found.

Which means NOT_FOUND in the scenario you describe, no?  No user means no schedule entry...

mykrupp, 2022-08-10:
You are technically correct. No user means no schedule entry. I didn't immediately jump to that conclusion. 

I guess I would have liked to clearly see something like 

NOT_FOUND If a valid user index is given, but no user is found in that index.

NOT_FOUND if both WeekDayIndex and UserIndex are within the valid range, however, there is not corresponding schedule entry found

Just my thought

bzbarsky-apple, 2022-08-10:
Yes, it would have been nice for the spec to be clearer here; a spec issue to that effect might be good.

In the meantime, the SDK needs to do what the spec says...  ;)

mykrupp, 2022-08-10:
The SDK should then probably also be doing the checking for missing schedules which is currently done in the linux example

bzbarsky-apple, 2023-02-13:
@mykrupp Was a spec issue ever raised here?

mykrupp, 2023-02-13:
I did not raise a spec issue. Willing to do so now if we still think this is relevant.

bzbarsky-apple, 2023-02-13:
@mykrupp Yes, this is still relevant....

mykrupp, 2023-02-13:
https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/6285

jrhees-cae, 2023-08-30:
This also similarly affects GetYearDaySchedule commands

PR #28974 (2023-09-01): [DRLK] Return NOT_FOUND error for non-existent User on GetWeekdaySchedule/GetYeardaySchedule commands
[DRLK] Return NOT_FOUND error for non-existent User on GetWeekdaySchedule/GetYeardaySchedule commands

- Updated YAML tests for TC-DRLK-2.5, TC-DRLK-2.7, DL_Schedules
- Updated door-lock-server

Fixes #21791

Corresponding spec updates: https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/7491
Corresponding test plan changes: https://github.com/CHIP-Specifications/chip-test-plans/pull/3395
commit 8bd7e087 (2023-09-01): [DRLK] Return NOT_FOUND error for non-existent User on GetWeekdaySchedule/GetYeardaySchedule commands (#28974)

* [DRLK] Return NOT_FOUND error for non-existent User on GetWeekdaySchedule/GetYeardaySchedule commands

Fixes #21791

* Restyled by prettier-yaml

* zap-regen

* Include changes to DL_Schedules.yaml test

* zap regen

* Fix issue with DL_Schedules for correct Status codes

---------
closed 2023-09-01 by mergify[bot]