project-chip/connectedhomeip #21795: Accessory is crashing when performing InvokeCommand with unsupported access.

krypton36, 2022-08-10:
#### Problem
When executing InvokeCommand on the accessory with Unsupported access, the accessory crashes.

Accessory SHA: aca4b94f9220f2dd23c44da5dbfcdf8f212f5e4d

[Accessory_log.txt](https://github.com/project-chip/connectedhomeip/files/9302087/Accessory_log.txt)

bzbarsky-apple, 2022-08-10:
On that SHA, I just tried:
1. Start all-clusters-app.
2. `chip-tool pairing code 17 749701123365521327694`
3. `chip-tool accesscontrol write acl '[]' 17 0`
4. `chip-tool onoff toggle 17 1`

and I get an UNSUPPORTED_ACCESS response, no crash...

bzbarsky-apple, 2022-08-10:
This crash was fixed in #21590.

The issue here is a client that is sending a Timed Request and then on the same exchange an _untimed_ invoke.  The server was detecting the mismatch, sending a status response (which closed the exchanged), then crashing in `CommandHandler::DecrementHoldOff` because it used the exchange pointer without null-checking or checking `mSentStatusResponse`.