project-chip/connectedhomeip #21814: Did not receive a response from TestStructArgumentRequest using Matter API

krypton36, 2022-08-11:
#### Problem
When invoking the command on the TestCluster(4294048773)-> TestStructArgumentRequest(7) using the API below in the Matter framework, we do not receive response data or get the command path object. No error was populated or no values were returned. The command was successful, but we expected a response.
```objective-c
- (void)invokeCommandWithEndpointId:(NSNumber *)endpointId
                          clusterId:(NSNumber *)clusterId
                          commandId:(NSNumber *)commandId
                      commandFields:(id)commandFields
                 timedInvokeTimeout:(NSNumber * _Nullable)timeoutMs
                        clientQueue:(dispatch_queue_t)clientQueue
                         completion:(MTRDeviceResponseHandler)completion;
```

For command field I have tried the following:
```json
{"type": "Structure", "value": [{"contextTag": 0, "data": {"type": "Structure", "value": [{"contextTag": 0, "data": {"type": "UnsignedInteger", "value": 5} }] } }]}
```

Note: chip-tool does receive the response data and is able to parse
./out/host/chip-tool any command-by-id 4294048773 0 {} 1122  7

```
[1660189054756] [16553:236922] CHIP: [DMG] Received Command Response Data, Endpoint=1 Cluster=0xFFF1_FC05 Command=0x0000_0008
[1660189054756] [16553:236922] CHIP: [TOO] Endpoint: 1 Cluster: 0xFFF1_FC05 Command 0x0000_0008
[1660189054757] [16553:236922] CHIP: [TOO]   BooleanResponse: {
[1660189054757] [16553:236922] CHIP: [TOO]     value: FALSE
[1660189054757] [16553:236922] CHIP: [TOO]    }
```

XML Data:
```xml
<command source="client" code="0x07" name="TestStructArgumentRequest"
             response="BooleanResponse" optional="true">
      <description>
        Command that takes an argument which is struct.  The response echoes the
        'b' field of the single arg.
      </description>
      <arg name="arg1" type="SimpleStruct"/>
</command>
```

Error occurs in darwin-framework-tool (a58afc505300b78feba5903b6e40b845f5561718):
`darwin-framework-tool any command-by-id 4294048773 7 {} 1122 1  `
```
[1660189292163] [16901:239752] CHIP: [TOO] Error: Error 0x0000008E
[1660189292163] [16901:239751] CHIP: [EM] Flushed pending ack for MessageCounter:233234467 on exchange 25370i
[1660189292163] [16901:239739] CHIP: [-] Error 0x0000008E at ../../commands/common/CHIPCommandBridge.mm:42
[1660189292163] [16901:239739] CHIP: [TOO] Run command failure: Error 0x0000008E
```

bzbarsky-apple, 2022-08-11:
>  ./out/host/chip-tool any command-by-id 4294048773 0 {} 1122 7

Shouldn't that be ` ./out/host/chip-tool any command-by-id 4294048773 7 {} 1122 1`?

bzbarsky-apple, 2022-08-11:
The bug here is in `NSObjectCommandCallback`.

In `NSObjectCommandCallback::OnResponse` it has:
```
    VerifyOrExit(aCommandPath.mClusterId == mClusterId && aCommandPath.mCommandId == mCommandId, err = CHIP_ERROR_SCHEMA_MISMATCH);
```
But `mCommandId` is the id of the command we _sent_.  This check will pass if the response is a status response, or if the request command happens to have the same id as the response command (which happens for some commands on some clusters), but in this case the request comand has id 7 and the response command has id 8.

In general, we have no way to know what the response command id will look like when we get a data response.

closed 2022-08-11 by bzbarsky-apple