project-chip/connectedhomeip #21834: Session resumption cache deletion partially failed for fabric index X

plan44, 2022-08-11:
#### Problem

Since the last HomePod beta update two days ago, the CHIP stack in the Homepod is not functional any more.

After successfully commissioning a device from an iPad, the DUT (bridge-app) enters operational mode and announces _matter._tcp, but is never connected by the HomePod.

Before, immediately after commissioning, the homepod started to issue IM requests to the bridge-app (repeated many times in the past weeks)

Retrieving the logs from the HomePod shows 47 times the following message in a row (all identical, all for fabric index 1):

```
error	2022-08-11 16:05:24.355540 +0200	homed	🔴 [1660226724356] [38:2084] CHIP: [SC] Session resumption cache deletion partially failed for fabric index 1, unable to load node state: /Library/Caches/com.apple.xbs/Sources/CHIPFramework/connectedhomeip/src/darwin/Framework/CHIP/MTRPersistentStorageDelega
```

and then a single

```
error	2022-08-11 16:05:25.056235 +0200	homed	🔴 [1660226725056] [38:2084] CHIP: [CTL] Warning, failed to delete session resumption state for fabric index 0x1: /Library/Caches/com.apple.xbs/Sources/CHIPFramework/connectedhomeip/src/darwin/Framework/CHIP/MTRPersistentStorageDelegateBridge.mm:43: CHIP Err
```

Not sure if these messages are related to the problem of not working as hub for matter accessories any more. As retrieving logs from the HomePod is very slow and often fails (AirDrop), I have little material to compare.

#### Steps tried to recover the homepod mini for testing
- removed homepod from home app, re-added anew -> does not help
- removed the Enable_matter profile and added again -> does not help

I can provide a full sysdiagnose snapshot on request.

bzbarsky-apple, 2022-08-11:
@kpark-apple @carol-apple Could you take a look please?

kpark-apple, 2022-08-11:
@plan44 Please provide full sysdiagnose snapshot.

bzbarsky-apple, 2022-08-11:
Fwiw, I suspect the session resumption bits are not the cause of the real problem, which is

> but is never connected by the HomePod.

though they could be another symptom of some underlying problem that causes the real issue.

plan44, 2022-08-11:
> @plan44 Please provide full sysdiagnose snapshot.

Posting here (210MB) is probably not a good idea? Where to send it to?

plan44, 2022-08-12:
In the meantime, I factory-reset the homepod with an 15.6 ipsw to make sure to get rid of corrupted data, then updated to 16 beta 5 again, and re-tested. **Still the homehub does not even start to connect the DUT after successful commissioning from the iPad**.

After several other attempts with an AppleTV I agree with @bzbarsky-apple that these session resumption errors are a red herring (no such errors happen on the AppleTV, but it also does not start communication with the DUT).

It seems to me that something has changed behind the scenes (outside connectedhomeip, in homed) with passing control to the home hubs after commissioning.

Neither current master, nor officially beta5 compatible aa9457e6b94b735076dff6297176183bf9780177 nor my last known working version between those get beyond the successful commissioning on the iPad, at the point when dns-sd advertising changes from `_matter._udp` to `_matter._tcp`, any more.

Also completely deleting the "home" and creating a new "home" did not change anything.

I'm aware that this is becoming quite OT for here - but I have no idea how to pinpoint the problem further.

[here](https://github.com/project-chip/connectedhomeip/files/9315190/homed.process.filtered.-.homepodmini.txt.zip) are all homed process log messages (extracted+filtered by console.app from system_logs.logarchive from the homepod). Let me know if you need more logs.

The only curious thing (besides the red herrings with session cache) I saw is:
```
info	2022-08-11 19:11:15.729926 +0200	homed	HAP2 Browser couldn't find Server with Identifier 0e:e0:f1:2d:13:8b. Maybe it's a CHIP accessory, checking...
info	2022-08-11 19:11:15.729970 +0200	homed	[CHIPAccessoryServerBrowser] discoverAccessoryServerWithIdentifier 0e:e0:f1:2d:13:8b
info	2022-08-11 19:11:15.730010 +0200	homed	[CHIPAccessoryServerBrowser] didn't find accessory server with identifier 0e:e0:f1:2d:13:8b
```

The AppleTV test later last night had something similar, but not equal:
```
error	23:53:03.019433+0200	homed	[6A53628D-81FB-45C4-85E2-915F9B4BC4F3/91:34:dd:83:3d:20+1/NO] Could not find CHIP accessory server to handle fetch CHIP pairings message
```

In both setups/homes, there are no accessories except for the matter DUT.

kpark-apple, 2022-08-12:
@plan44 Thank you for the log. There was a mismatch of the state stored from phone vs check in HomePod. We'll work on a fix.

```
info	2022-08-11 19:32:58.737360 +0200	homed	[CHIPAccessoryServerBrowser] One or more node parameters are unavailable; deferring pairing setup to a later point when all parameters are available
```

plan44, 2022-08-26:
> @kpark-apple [...] We'll work on a fix.

Apparently, the fix made it to the beta released last Tuesday 😄 👍Thanks!

closed 2022-08-26 by plan44