project-chip/connectedhomeip #21835: VerifyOrDie failure during minimalmdns operational discovery

s-mcclain, 2022-08-11:
#### Problem
* Having the chip-tool repeatedly send out cluster commands to another device, every so often (between 50-150) during operational discovery, once the controller sends out its mdns query, almost immediately afterwards, a VerifyOrDie SIGABRT is thrown during one of two areas:
  * VerifyOrDie failure at vendor/csa-iot/connectedhomeip/src/lib/support/Variant.h:217: (mTypeId == VariantInternal::TupleIndexOfType<T, std::tuple<Ts...>>::value)
  * VerifyOrDie failure at connectedhomeip/src/lib/support/Pool.cpp:71: (value & (kBit1 << offset)) != 0
 
* This seems to occur during the Node resolution / parsing of the mdns query responses. Possibly that the buffers for these packets are getting freed while still being parsed?

Below are the call stacks for the two different areas of the SIGABRT:

```
08-09 11:52:55.884  6332  6332 D CHIP SPT: VerifyOrDie failure at connectedhomeip/src/lib/support/Pool.cpp:71: (value & (kBit1 << offset)) != 0
08-09 11:52:55.884  6332  6332 F libc    : Fatal signal 6 (SIGABRT), code -6 in tid 6332 (chip_tool_client)
08-09 11:52:55.885   200   200 W         : debuggerd: handling request: pid=6332 uid=0 gid=0 tid=6332
08-09 11:52:55.943  6335  6335 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
08-09 11:52:55.943  6335  6335 F DEBUG   : Build fingerprint: 'Amazon/octave/octave:6.0/NMAIN1/3706N:user/amz-p,release-keys'
08-09 11:52:55.943  6335  6335 F DEBUG   : Revision: '0'
08-09 11:52:55.943  6335  6335 F DEBUG   : ABI: 'arm'
08-09 11:52:55.943  6335  6335 F DEBUG   : pid: 6332, tid: 6332, oom: 0, name: chip_tool_client
08-09 11:52:55.943  6335  6335 F DEBUG   : signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
08-09 11:52:55.943  6335  6335 F DEBUG   :     r0 00000000  r1 000018bc  r2 00000006  r3 00000008
08-09 11:52:55.943  6335  6335 F DEBUG   :     r4 e7ef658c  r5 00000006  r6 e7ef6534  r7 0000010c
08-09 11:52:55.943  6335  6335 F DEBUG   :     r8 ffffffff  r9 00000000  sl e7cb0600  fp ffca60b0
08-09 11:52:55.943  6335  6335 F DEBUG   :     ip 00000001  sp ffca5f78  lr e792c4a7  pc e792ed04  cpsr 200f0010
08-09 11:52:55.948  6335  6335 F DEBUG   : 
08-09 11:52:55.948  6335  6335 F DEBUG   : backtrace:
08-09 11:52:55.948  6335  6335 F DEBUG   :     #00 pc 00049d04  /system/lib/libc.so (tgkill+12)
08-09 11:52:55.948  6335  6335 F DEBUG   :     #01 pc 000474a3  /system/lib/libc.so (pthread_kill+34)
08-09 11:52:55.948  6335  6335 F DEBUG   :     #02 pc 0001d765  /system/lib/libc.so (raise+10)
08-09 11:52:55.948  6335  6335 F DEBUG   :     #03 pc 000192b1  /system/lib/libc.so (__libc_android_abort+34)
08-09 11:52:55.948  6335  6335 F DEBUG   :     #04 pc 00017270  /system/lib/libc.so (abort+4)
08-09 11:52:55.948  6335  6335 F DEBUG   :     #05 pc 0012eb0b  /system/lib/libchip_tool_client.so
08-09 11:52:55.948  6335  6335 F DEBUG   :     #06 pc 0015adcb  /system/lib/libchip_tool_client.so (_ZN4chip8internal21StaticAllocatorBitmap10DeallocateEPv+106)
08-09 11:52:55.948  6335  6335 F DEBUG   :     #07 pc 0019d96b  /system/lib/libchip_tool_client.so (_ZN4chip6System15LayerImplSelect11CancelTimerEPFvPNS0_5LayerEPvES4_+42)
08-09 11:52:55.948  6335  6335 F DEBUG   :     #08 pc 00154103  /system/lib/libchip_tool_client.so
08-09 11:52:55.948  6335  6335 F DEBUG   :     #09 pc 0015404f  /system/lib/libchip_tool_client.so
08-09 11:52:55.948  6335  6335 F DEBUG   :     #10 pc 001503ef  /system/lib/libchip_tool_client.so (_ZN4chip14AddressResolve4Impl8Resolver10LookupNodeERKNS0_17NodeLookupRequestERNS1_16NodeLookupHandleE+166)
08-09 11:52:55.948  6335  6335 F DEBUG   :     #11 pc 001370f9  /system/lib/libchip_tool_client.so (_ZN4chip22OperationalDeviceProxy17LookupPeerAddressEv+92)
08-09 11:52:55.948  6335  6335 F DEBUG   :     #12 pc 00136fc9  /system/lib/libchip_tool_client.so (_ZN4chip22OperationalDeviceProxy7ConnectEPNS_8Callback8CallbackIPFvPvPS0_EEEPNS2_IPFvS3_NS_6PeerIdENS_9ChipErrorEEEE+44)
08-09 11:52:55.948  6335  6335 F DEBUG   :     #13 pc 00132ea1  /system/lib/libchip_tool_client.so (_ZN4chip18CASESessionManager22FindOrEstablishSessionENS_6PeerIdEPNS_8Callback8CallbackIPFvPvPNS_22OperationalDeviceProxyEEEEPNS3_IPFvS4_S1_NS_9ChipErrorEEEE+104)
08-09 11:52:55.948  6335  6335 F DEBUG   :     #14 pc 000fb287  /system/lib/libchip_tool_client.so
08-09 11:52:55.948  6335  6335 F DEBUG   :     #15 pc 000f3d85  /system/lib/libchip_tool_client.so (chip_clusterSendCommand+5400)
08-09 11:52:55.948  6335  6335 F DEBUG   :     #16 pc 00004ed9  /system/bin/chip_tool_client_test
08-09 11:52:55.948  6335  6335 F DEBUG   :     #17 pc 00007da9  /system/bin/chip_tool_client_test
08-09 11:52:55.948  6335  6335 F DEBUG   :     #18 pc 0000acc3  /system/bin/chip_tool_client_test
08-09 11:52:55.948  6335  6335 F DEBUG   :     #19 pc 0000afa9  /system/bin/chip_tool_client_test
08-09 11:52:55.948  6335  6335 F DEBUG   :     #20 pc 00016e89  /system/lib/libc.so (__libc_init+48)
08-09 11:52:55.948  6335  6335 F DEBUG   :     #21 pc 00001720  /system/bin/chip_tool_client_test
```

```
08-11 04:25:23.746  6765  6765 D CHIP CSM: FindOrEstablishSession: PeerId = 441460FAD3F48F0A:00000000000004D2
08-11 04:25:23.746  6765  6765 D CHIP CSM: FindOrEstablishSession: No existing OperationalDeviceProxy instance found
08-11 04:25:23.746  6765  6765 D CHIP CTL: OperationalDeviceProxy[441460FAD3F48F0A:00000000000004D2]: State change 1 --> 2
08-11 04:25:23.748  6765  6767 D CHIP SPT: VerifyOrDie failure at vendor/csa-iot/connectedhomeip/src/lib/support/Variant.h:217: (mTypeId == VariantInternal::TupleIndexOfType<T, std::tuple<Ts...>>::value)
08-11 04:25:23.749  6765  6767 F libc    : Fatal signal 6 (SIGABRT), code -6 in tid 6767 (chip_tool_client)
08-11 04:25:23.752   226   226 W         : debuggerd: handling request: pid=6765 uid=0 gid=0 tid=6767
08-11 04:25:23.813  6768  6768 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
08-11 04:25:23.814  6768  6768 F DEBUG   : Build fingerprint: 'Amazon/octave/octave:6.0/NMAIN1/60460P:userdebug/amz-d,dev-keys'
08-11 04:25:23.814  6768  6768 F DEBUG   : Revision: '0'
08-11 04:25:23.814  6768  6768 F DEBUG   : ABI: 'arm'
08-11 04:25:23.814  6768  6768 F DEBUG   : pid: 6765, tid: 6767, oom: 0, name: chip_tool_client  >>> chip_tool_client_test <<<
08-11 04:25:23.814  6768  6768 F DEBUG   : signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
08-11 04:25:23.814  6768  6768 F DEBUG   :     r0 00000000  r1 00001a6f  r2 00000006  r3 00000008
08-11 04:25:23.814  6768  6768 F DEBUG   :     r4 ee67f978  r5 00000006  r6 ee67f920  r7 0000010c
08-11 04:25:23.814  6768  6768 F DEBUG   :     r8 eea4e600  r9 eea4e090  sl 00000000  fp ee67f458
08-11 04:25:23.814  6768  6768 F DEBUG   :     ip 00000000  sp ee67f360  lr eeab74a7  pc eeab9d04  cpsr 200f0010
08-11 04:25:23.826  6768  6768 F DEBUG   : 
08-11 04:25:23.826  6768  6768 F DEBUG   : backtrace:
08-11 04:25:23.826  6768  6768 F DEBUG   :     #00 pc 00049d04  /system/lib/libc.so (tgkill+12)
08-11 04:25:23.826  6768  6768 F DEBUG   :     #01 pc 000474a3  /system/lib/libc.so (pthread_kill+34)
08-11 04:25:23.826  6768  6768 F DEBUG   :     #02 pc 0001d765  /system/lib/libc.so (raise+10)
08-11 04:25:23.826  6768  6768 F DEBUG   :     #03 pc 000192b1  /system/lib/libc.so (__libc_android_abort+34)
08-11 04:25:23.826  6768  6768 F DEBUG   :     #04 pc 00017270  /system/lib/libc.so (abort+4)
08-11 04:25:23.826  6768  6768 F DEBUG   :     #05 pc 0012eb0b  /system/lib/libchip_tool_client.so
08-11 04:25:23.826  6768  6768 F DEBUG   :     #06 pc 0019aff7  /system/lib/libchip_tool_client.so
08-11 04:25:23.826  6768  6768 F DEBUG   :     #07 pc 0019b3c3  /system/lib/libchip_tool_client.so (_ZN4chip5Dnssd19IncrementalResolver4TakeERNS0_16ResolvedNodeDataE+70)
08-11 04:25:23.826  6768  6768 F DEBUG   :     #08 pc 0015388f  /system/lib/libchip_tool_client.so
08-11 04:25:23.826  6768  6768 F DEBUG   :     #09 pc 00155465  /system/lib/libchip_tool_client.so (_ZN4mdns7Minimal10ServerBase19OnUdpPacketReceivedEPN4chip4Inet11UDPEndPointEONS2_6System18PacketBufferHandleEPKNS3_12IPPacketInfoE+84)
08-11 04:25:23.826  6768  6768 F DEBUG   :     #10 pc 001975fb  /system/lib/libchip_tool_client.so (_ZN4chip4Inet22UDPEndPointImplSockets15HandlePendingIOENS_8BitFlagsINS_6System16SocketEventFlagsEhEE+422)
08-11 04:25:23.826  6768  6768 F DEBUG   :     #11 pc 0019dd83  /system/lib/libchip_tool_client.so (_ZN4chip6System15LayerImplSelect12HandleEventsEv+202)
[... cut ...]
08-11 04:25:23.827  6768  6768 F DEBUG   :     #14 pc 00046f73  /system/lib/libc.so (_ZL15__pthread_startPv+22)
08-11 04:25:23.827  6768  6768 F DEBUG   :     #15 pc 00019cfd  /system/lib/libc.so (__start_thread+6)
```

* System Configuration:
  * A chip-tool controller that uses minimalmdns server with ephemeral ports enabled, as well as advertises itself as operational during server startup.

 #### Proposed Solution
Does the Matter SDK allocate and copy their own buffers while parsing the mdns query?
Is the MinimalMdnsServer thread safe while parsing the receiving query responses?

s-mcclain, 2022-08-12:
[tcpdump.zip](https://github.com/project-chip/connectedhomeip/files/9312905/tcpdump.zip)
Attached is the tcpdump of when the SIGABRT actually occurs.
Will try to get the full call stack tomorrow.

s-mcclain, 2022-08-12:
So I was able to confirm after a little more debugging that:

Inside chip::Dnssd::(anonymous namespace)::MinMdnsResolver::AdvancePendingResolverStates(), we get into the 'resolver->IsActiveOperationalParse()' scope. At the start, the resolver->IsActive == true.
Somewhere while resolver->Take() is being called, the function "chip::Dnssd::IncrementalResolver::ResetToInactive()" gets called, where it resets the mSpecificResData. So when Take() gets called, which calls "mSpecificResolutionData.Get<OperationalNodeData>();", the mSpecificResolutionData.mTypeId is set to -1 (kInvalidType.

My logs that print out to help debug look something like this:

```
01-01 01:58:14.960 24096 24098 I CHIP DIS: static void mdns::Minimal::ServerBase::OnUdpPacketReceived(chip::Inet::UDPEndPoint *, chip::System::PacketBufferHandle &&, const chip::Inet::IPPacketInfo *): We have received on endpoint: 0x0xaff78c20
01-01 01:58:14.960 24096 24098 I CHIP DL : virtual void chip::Dnssd::(anonymous namespace)::MinMdnsResolver::OnMdnsPacketData(const mdns::Minimal::BytesRange &, const chip::Inet::IPPacketInfo *): BytesRange is 0x0xaf47f664
01-01 01:58:14.960 24096 24098 I CHIP DL : void chip::Dnssd::(anonymous namespace)::MinMdnsResolver::AdvancePendingResolverStates(): Resolver ptr: 0xaff761a0. isactive? (1)
01-01 01:58:14.960 24096 24098 I CHIP DL : void chip::Dnssd::(anonymous namespace)::MinMdnsResolver::AdvancePendingResolverStates():L390: ResolverIsActiveOperationalParse(). ResolvedNodeData: 0xaf47f450. resolver: 0xaff761a0. isactive?: 1
01-01 01:58:14.960 24096 24098 I CHIP DL : CHIP_ERROR chip::Dnssd::IncrementalResolver::Take(chip::Dnssd::ResolvedNodeData &): mSpecificResolutionData: 0xaff762a0, type=0
01-01 01:58:14.960 24096 24098 I CHIP DL : void chip::Dnssd::IncrementalResolver::ResetToInactive(): Resetting 0xaff761a0 (mSpecificResData=0xaff762a0)
01-01 01:58:14.960 24096 24098 I CHIP DIS: static void mdns::Minimal::ServerBase::OnUdpPacketReceived(chip::Inet::UDPEndPoint *, chip::System::PacketBufferHandle &&, const chip::Inet::IPPacketInfo *): We have received on endpoint: 0x0xaff78c80
01-01 01:58:14.961 24096 24098 I CHIP DL : virtual void chip::Dnssd::(anonymous namespace)::MinMdnsResolver::OnMdnsPacketData(const mdns::Minimal::BytesRange &, const chip::Inet::IPPacketInfo *): BytesRange is 0x0xaf47f664
01-01 01:58:14.961 24096 24098 I CHIP DL : void chip::Dnssd::(anonymous namespace)::MinMdnsResolver::AdvancePendingResolverStates(): Resolver ptr: 0xaff761a0. isactive? (1)
01-01 01:58:14.961 24096 24096 I CHIP DIS: mDNS broadcast had only partial success: 2 successes and 2 failures.
01-01 01:58:14.961 24096 24098 I CHIP DL : void chip::Dnssd::(anonymous namespace)::MinMdnsResolver::AdvancePendingResolverStates():L390: ResolverIsActiveOperationalParse(). ResolvedNodeData: 0xaf47f450. resolver: 0xaff761a0. isactive?: 1
01-01 01:58:14.961 24096 24096 I CHIP DL : void chip::Dnssd::IncrementalResolver::ResetToInactive(): Resetting 0xaff761a0 (mSpecificResData=0xaff762a0)
01-01 01:58:14.961 24096 24098 I CHIP DL : CHIP_ERROR chip::Dnssd::IncrementalResolver::Take(chip::Dnssd::ResolvedNodeData &): mSpecificResolutionData: 0xaff762a0, type=0
01-01 01:58:14.961 24096 24098 I CHIP DL : T &chip::Variant<chip::Dnssd::OperationalNodeData, chip::Dnssd::CommissionNodeData>::Get() [Ts = <chip::Dnssd::OperationalNodeData, chip::Dnssd::CommissionNodeData>, T = chip::Dnssd::OperationalNodeData]: mTypeId (-1) != 0 | Valid? (0)
01-01 01:58:14.961 24096 24098 I CHIP SPT: VerifyOrDie failure at vendor/csa-iot/connectedhomeip/src/lib/support/Variant.h:221: false
```

s-mcclain, 2022-08-13:
Looks like it's getting expired by the function:
`ExpireIncrementalResolvers()`

```
01-01 02:27:31.288 29666 29668 I CHIP DL : CHIP_ERROR chip::Dnssd::IncrementalResolver::Take(chip::Dnssd::ResolvedNodeData &): resolver (0xaad1b1a0) mSpecificResolutionData: 0xaad1b2a0, type=0
01-01 02:27:31.288 29666 29666 I CHIP DL : void chip::Dnssd::(anonymous namespace)::MinMdnsResolver::ExpireIncrementalResolvers(): Resetting resolver (0xaad1b1a0) to inactive
01-01 02:27:31.288 29666 29666 I CHIP DL : void chip::Dnssd::IncrementalResolver::ResetToInactive(): Resetting 0xaad1b1a0 (mSpecificResData=0xaad1b2a0)
01-01 02:27:31.288 29666 29668 I CHIP DL : T &chip::Variant<chip::Dnssd::OperationalNodeData, chip::Dnssd::CommissionNodeData>::Get() [Ts = <chip::Dnssd::OperationalNodeData, chip::Dnssd::CommissionNodeData>, T = chip::Dnssd::OperationalNodeData]: mTypeId (-1) != 0 | Valid? (0)
```

s-mcclain, 2022-08-15:
@andy31415 / @andreilitvin Think you'd be able to help assist in how we can mitigate this during active parsing/resolution?

andy31415, 2022-08-17:
Is there some form of script example on how to reproduce this? The `chip-tool repeatedly send out cluster commands to another device` is not as clear to me ... do I just loop a `readattribute` or something like that?

andy31415, 2022-08-17:
Regarding the thread safety and allocation questions: the chip stack generally assumes a single thread for data processing. in particular, minmdns assumes that packets are received and parsed in-line by the "chip thread" and I do not expect it to manage its own allocation/deallocation. 

it does however allocate extra buffers if it needs to reply to packets.

andy31415, 2022-08-17:
For https://github.com/project-chip/connectedhomeip/issues/21835#issuecomment-1213595012 - does this mean that you observe two separate threads interacting/racing?

andy31415, 2022-08-17:
In terms of how things should work: `ExpireIncrementalResolvers` is supposed to mark some resolvers as forcibly done (i.e. timeout before being able to find something), which would expect to get their timeout called and operations should terminate.

It should not result in a double-free which the verify_or_die seems to indicate. Maybe enabling pools on the heap and some memory sanitize could help narrow down where each individual free happens (assuming this pool can be moved to heap).

andy31415, 2022-08-17:
From what I can see, `ExpireIncrementalResolvers` is supposed to check if any resolver still needs data and if all needed data is received (i.e. they have an IP) they they are marked as complete.

s-mcclain, 2022-08-17:
@andy31415 Yes we basically have a script which repeats over the following commands 100+ times:
```
chip-tool colorcontrol move-to-hue 0 7 0 0 0 1234 1
chip-tool colorcontrol move-to-saturation 3 0 0 0 1234 1
chip-tool levelcontrol move-to-level 100 0 0 0 1234 1
```

So would we want to be able to add a sort of "state" to the resolver to indicate whether it's still being parsed within the AdvancePendingResolverStates or not?

andy31415, 2022-08-17:
What I do not quite understand is:

```
VerifyOrDie failure at vendor/csa-iot/connectedhomeip/src/lib/support/Variant.h:221:
```

seems to indicate that the object we call `Get` on is not the right type. However the code has checks like `VerifyOrReturnError(IsActiveOperationalParse(), CHIP_ERROR_INCORRECT_STATE);`  which is effecively a:

```cpp
bool IsActiveOperationalParse() const { return mSpecificResolutionData.Is<OperationalNodeData>(); }
```

so that Get should not fail given that the check before it succeeds... this is unless there is some multi-threading going on (which should not happen).

Also the original error is `VerifyOrDie failure at connectedhomeip/src/lib/support/Pool.cpp:71: (value & (kBit1 << offset)) != 0` which seems to be something else - it is Pool.cpp and not Variant.

I believe I would need to reproduce it myself ideally or maybe if you are on slack see if we get some time overlap so we can debug on your setup. Just reading the code right now does not make anything stand out.

andy31415, 2022-08-17:
Is this a custom build of sorts or chip-tool? I wonder if some threading issues exist - e.g. some way of chip main thread being started several times. Maybe printing the thread-id during diferent operations may show when take() is called and by what. This behaviour is very odd.

s-mcclain, 2022-08-17:
@andy31415  For sure, I'm on slack. I'll try to reach out to you Friday. Meanwhile, I'll check on the thread ids to see what thread(s) are calling these commands.

s-mcclain, 2022-08-18:
@andy31415  In the logs I posted, the actual Thread ID is included for each line:
```
// Format: <date> <time> <processid> <tid> | <log>
// This happens on ThreadID 29668
01-01 02:27:31.288 29666 29668 I CHIP DL : CHIP_ERROR chip::Dnssd::IncrementalResolver::Take(chip::Dnssd::ResolvedNodeData &): resolver (0xaad1b1a0) mSpecificResolutionData: 0xaad1b2a0, type=0
// This happens on ThreadID 29666
01-01 02:27:31.288 29666 29666 I CHIP DL : void chip::Dnssd::(anonymous namespace)::MinMdnsResolver::ExpireIncrementalResolvers(): Resetting resolver (0xaad1b1a0) to inactive
01-01 02:27:31.288 29666 29666 I CHIP DL : void chip::Dnssd::IncrementalResolver::ResetToInactive(): Resetting 0xaad1b1a0 (mSpecificResData=0xaad1b2a0)
01-01 02:27:31.288 29666 29668 I CHIP DL : T &chip::Variant<chip::Dnssd::OperationalNodeData, chip::Dnssd::CommissionNodeData>::Get() [Ts = <chip::Dnssd::OperationalNodeData, chip::Dnssd::CommissionNodeData>, T = chip::Dnssd::OperationalNodeData]: mTypeId (-1) != 0 | Valid? (0)
```

The CHIP task thread is 29668. So you're right that there's multiple threads. Is this not allowed to have the initialization done within the main process?

andy31415, 2022-08-18:
I am unsure - we can do things in main() however based on timestamps it looks like both operations happen at the same time (31.288) so we have two threads operating on the same dnssd processing logic. This is generally unexpected as CHIP is designed to work as a single thread. 

If you could find a way to get the stack trace of each thread that would help understand things. the main() thread is allowed to do chip operations, however only _before_ the chip main loop is started. You could also try to add some `assertChipStackLockedByCurrentThread` in all various incremental resolver places (like in ::Take and ::ResetToInactive) and figure out if that asserts. if it does, we likely found the issue. If it does not assert, I still wonder what those threads are.

s-mcclain, 2022-08-19:
@andy31415 Because we're setting `enableServerInteractions` to true and initializing the DeviceControllerFactory before the chip main loop starts, this would startup a DNS-SD server outside of the CHIP thread. Then when we call DeviceControllerFactory.SetupCommissioner() this seems to start the DNS-SD server in the CHIP thread. Which looks like this is where the race condition is from, as we have two servers starting up almost immediately after each other.

andy31415, 2022-08-19:
This seems like a bug - there should be a single DNS-SD server running at any time. I believe the dnssd server should be a singleton so even though it has a `StartServer` method, that should only restart the same server and not use separate threads.

Could you clarify the statement of `we have two servers starting`? Is this actually two servers (different `this`) or is it two different threads (main and chip thread) conflicting? I would expect the latter.

s-mcclain, 2022-08-19:
Sorry for confusion, yes two different threads (main and chip thread) which are currently conflicting with each other. This conflict only occurs when the enableServerInteractions is enabled.

andy31415, 2022-08-19:
What is the sequence of operations in terms of using `serverInteractions` and starting the chip main thread?

Things should be safe to run before chip main thread is started (i.e. setting enableServerInteractions), however once you start the chip thread, you are not allowed to operate on anything else from main. So if you do:

```
ControllerFactory::..... 
StartChipMainThread
```

that is safe (I believe), however:

```
StartChipMainThread()
ControllerFactory::....
```

is not ok.

s-mcclain, 2022-08-24:
@andy31415 Thanks for that example. We were doing several things after calling the DeviceController::Init, which may have queued up a number of tasks for the thread before starting the thread that created a race condition. After fixing this, I do not see the reported SIGABRT occurring any more.

closed 2022-08-24 by s-mcclain