project-chip/connectedhomeip #21847: TH failed to change the attribute value for privious 2 subscribed attributes when  subscription are active for three atributes

Rajashreekalmane, 2022-08-12:
#### Problem
 When TH send the 3 subscription request to DUT. Each subscribe request should contain 3 different paths.
 Once all subscriptions are active(all Subscription Requests succeed.),  it shouldn't cancel the previous subscription  when TH try to change the value of all the attributes that have been subscribed or trigger an action on the DUT to generate an event.but currently it is canceling   the previous 2 subscription and getting report data with appropriate attribute value for last subscription.

 **expected behavior:-** TH should change the value of all 3 attributes that have been subscribed on DUT
 *** actual behavior:-** TH  it is canceling  the previous 2 subscription and getting report data with appropriate attribute value for last subscription.

 *** steps to reproduce:-**

1.   ./chip-tool interactive start
2.  onoff  subscribe on-time  100 1000  1 1
3.  levelcontrol  subscribe  on-level 100 1000  1 1
4. onoff subscribe start-up-on-off   100 1000 1 1
Once all above subscription are active try to change the attribute value by sending below commands
5. onoff  write on-time 1 1 1
6.  levelcontrol  write  on-level  1  1 1
7.  onoff write start-up-on-off  1 1 1

**system configuration**
Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM
Platform : IP and wifi
Commit-ID : https://github.com/project-chip/connectedhomeip/commit/a1643eee5ef5534d494ae205bb586b4a9cbfee23
Dut application: all-clusters-app

**Test-Plan reference:-**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-7-1-multi-fabric-subscription-test-cases-dut_server

**PFA log below:-**
[[TC-IDM-7.1]Subscription.txt](https://github.com/project-chip/connectedhomeip/files/9314962/TC-IDM-7.1.Subscription.txt)

bzbarsky-apple, 2022-08-12:
@Rajashreekalmane Those commands are not sending `--keepSubscriptions true` with the subscriptions, so each one cancels all the previous ones.

bzbarsky-apple, 2022-08-12:
@cjandhyala We may need changes to the verification steps here.

closed 2022-08-12 by bzbarsky-apple