project-chip/connectedhomeip #21850: [EFR32] Issue while running OTA test case

rosahay-silabs, 2022-08-12:
#### Problem
 * expected behavior
    OTA updates should load correctly
 * actual behavior
    The process fails with `00> <error > [SWU] ERROR: In HandleProcessBlock bootloader_eraseWriteStorage() error 1028`
 * steps to reproduce
    ```shell
    rm -rf /tmp/chip_*
    chip-tool pairing ble-wifi 21 <SSID> <KEY> 20202021 3842
    screen -dmS chip-ota-provider ./out/debug/chip-ota-provider-app -f ./out/lighting-app/BRD4161A/chip-efr32-lighting-example.ota
    chip-tool pairing onnetwork 1 20202021
    chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"fabricIndex": 1, "privilege": 3, "authMode": 2, "subjects": null, "targets": null}]' 1 0
    chip-tool otasoftwareupdaterequestor announce-ota-provider 1 0 0 0 21 0
    ````
 * system configuration
    EFR32MG12 + RS9116 (device under test), Raspberry Pi 4
 * screenshots, images, logs, etc.

rosahay-silabs, 2022-08-12:
attaching [ota-failure.log](https://github.com/project-chip/connectedhomeip/files/9315642/ota-failure.log)

selissia, 2022-08-12:
This error from the attached log
```<error > [SWU] ERROR: In HandleProcessBlock bootloader_eraseWriteStorage() error 1025```
i.e. `BOOTLOADER_ERROR_STORAGE_INVALID_SLOT` suggests that the Storage Slot is not set up correctly -- this in turn suggests that the device is not running a bootloader built for Matter OTA from `third_party/silabs/matter_support/matter/efr32/bootloader_binaries` as described in https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/silabs_efr32_software_update.md

rosahay-silabs, 2022-08-12:
> 

> `BOOTLOADER_ERROR_STORAGE_INVALID_SLOT`

Hey @selissia, 
I am facing this issue despite using the bootloader from the specified location.

rosahay-silabs, 2022-08-16:
Hey @selissia 
Attaching the logs from the chip tool and OTA provider

[chip-tool.run.log](https://github.com/project-chip/connectedhomeip/files/9349227/chip-tool.run.log)

[chip-ota-provider.run.log](https://github.com/project-chip/connectedhomeip/files/9349226/chip-ota-provider.run.log)

selissia, 2022-08-16:
@chirag-silabs @rosahay-silabs Please provide the name and the size of the bootloader binary (the .s37 file) that you are using.

selissia, 2022-08-16:
I have tested the doorlock app over Thread on BRD4161A with 5033def30aadf96cf84188b6dac5a5a122b772fc with the following bootloader and OTA update works.  
bootloader-storage-spiflash-single-1024k-BRD4161A-gsdk4.1.s37 , file size 53956

chirag-silabs, 2022-08-17:
I have tested the door lock over Wi-Fi on BRD4162A with bootloader-storage-spiflash-single-1024k-BRD4162A-gsdk4.1.s37, file size 53KB.

rosahay-silabs, 2022-08-19:
@selissia, there seems to be an issue with BRD4162A, I am using [bootloader-storage-spiflash-single-1024k-BRD4162A-gsdk4.1.s37](https://github.com/SiliconLabs/sdk_support/blob/main/matter/efr32/bootloader_binaries/bootloader-storage-spiflash-single-1024k-BRD4162A-gsdk4.1.s37)

closed 2022-10-11 by rosahay-silabs