project-chip/connectedhomeip #21853: [Raspberry Pi] Cross compile chip-tool for raspberry pi

rosahay-silabs, 2022-08-12:
#### Problem
 * expected behavior
    Developer should be able to cross compile chip-tool and other utilities on another host machine.
 * actual behavior
    Cross compilation fails on the `ld` step.
 * steps to reproduce
    ```shell
    # setup sysroot
    cd /opt/raspbian
    mkdir sysroot
    cd sysroot

    rsync -avz pi@pi.local:/lib .
    rsync -avz pi@pi.local:/usr/include usr
    rsync -avz pi@pi.local:/usr/lib usr
    ```

    ```shell
    cd example/chip-tool
    gn gen --args='target_cpu="arm64" target_os="linux" sysroot="/opt/raspbian/sysroot" system_libdir="lib/aarch64-linux-gnu" is_clang=true' out/debug
    ```
 * system configuration
    WSL2 UBUNTU 20.04LTS on Windows 10, sysroot from Raspbian
 * screenshots, images, logs, etc.
    ```shell
    ninja: Entering directory `out/debug'
    [508/509] ld ./chip-tool
    FAILED: chip-tool chip-tool.map
    clang++ --target=aarch64-linux-gnu -O0 -fPIC -Werror -Wl,--fatal-warnings -Wl,-z,defs -fdiagnostics-color --sysroot=/opt/raspbian/sysroot/ -Wl,--gc-sections -pie -L/opt/raspbian/sysroot/usr/lib/aarch64-linux-gnu -Wl,-Map,./chip-tool.map @./chip-tool.rsp -o ./chip-tool
    ld.lld: error: undefined symbol: _dl_stack_flags
    >>> referenced by allocatestack.c:525
    >>>               pthread_create.o:(__pthread_create_2_1) in archive /opt/raspbian/sysroot/usr/lib/aarch64-linux-gnu/libpthread.a
    >>> referenced by allocatestack.c:647
    >>>               pthread_create.o:(__pthread_create_2_1) in archive /opt/raspbian/sysroot/usr/lib/aarch64-linux-gnu/libpthread.a

    ld.lld: error: undefined symbol: __pointer_chk_guard_local
    >>> referenced by jmpbuf-offsets.h:52 (../sysdeps/aarch64/jmpbuf-offsets.h:52)
    >>>               unwind.o:(unwind_stop) in archive /opt/raspbian/sysroot/usr/lib/aarch64-linux-gnu/libpthread.a

    ld.lld: error: undefined symbol: __dladdr
    >>> referenced by dladdr.o:(dladdr) in archive /opt/raspbian/sysroot/usr/lib/aarch64-linux-gnu/libdl.a
    clang-14: error: linker command failed with exit code 1 (use -v to see invocation)
    ninja: build stopped: subcommand failed.
    ```
    
    ```shell
    $ find . -name libpthread.so
    ./usr/lib/aarch64-linux-gnu/libpthread.so
    $ find . -name libpthread.a
    ./usr/lib/aarch64-linux-gnu/libpthread.a
    ```

bzbarsky-apple, 2022-08-12:
@mspang could you take a look, please?

mspang, 2022-08-25:
This is not a supported workflow.

Is libpthread.so a broken link?

e.g. on ubuntu

```
ubuntu@ubuntu:/usr/lib/aarch64-linux-gnu$ ls -l libpthread.so
lrwxrwxrwx 1 root root 38 Mar 31  2021 libpthread.so -> /lib/aarch64-linux-gnu/libpthread.so.0
```

Your sysroot is broken.

mspang, 2022-08-25:
Hint: libpthread.so is an absolute link, and you've moved the link off the system where the link target exists.

mspang, 2022-08-25:
You should be able to build using the canned arm64 linux sysroot (based on ubuntu, but it shouldn't matter, at least as long as raspbian is reasonably up to date),

https://chrome-infra-packages.appspot.com/p/experimental/matter/sysroot/ubuntu-21.04-aarch64/+/sumaA20JcnhLrFYO7glg1tfeKQWA4Khmquv5IlME2IMC

Our CI has been doing this since last year. You don't need to build against the exact distro.

mspang, 2022-08-25:
You could also try rsync -L

closed 2022-08-25 by mspang

rosahay-silabs, 2022-09-14:
I was able to fix the symlinks and compile for raspbian
 ### SETUP
```shell
# setup sysroot
mkdir -p /opt/raspbian-bullseye-aarch64-sysroot/
cd /opt/raspbian-bullseye-aarch64-sysroot/

rsync -avz pi@pi.local:/lib .
rsync -avz pi@pi.local:/usr/include usr
rsync -avz pi@pi.local:/usr/lib usr
```
#### FIX
The following fixes the symlinks

```shell
find /opt/raspbian-bullseye-aarch64-sysroot/ -type l -lname '/lib/*' -printf 'ln -nsf "$(readlink "%p" | sed s/lib/opt\\\\/raspbian-bullseye-aarch64-sysroot\\\\/lib/)" "%p"\n' > script.sh

sh ./script.sh
```

Now I can compile the chip-tool on my WSL2 ubuntu without have to clone the codebase on the SD card.