project-chip/connectedhomeip #21863: Status Response not being sent when isTimed flag is set incorrectly

mrjerryjohns, 2022-08-12:
#### Problem

In `TestCommandInteraction::TestCommandHandlerWithProcessReceivedEmptyDataMsg`, the act of sending an invoke where messageIsTimed did not match transactionIsTimed previously used to result in a status response being sent out.

I don't see that anymore after I synced up to master

bzbarsky-apple, 2022-08-12:
In a normal message reception flow it would be sent (by the caller).  TestCommandHandlerWithProcessReceivedEmptyDataMsg directly calls an internal implementation message that returns status that the caller then sends.   Perhaps it should check the returned status?

mrjerryjohns, 2022-08-12:
Ah ok. In that case, will take off the spec label.

yunhanw-google, 2022-08-12:
https://github.com/project-chip/connectedhomeip/pull/21871

yunhanw-google, 2022-08-15:
fixed

closed 2022-08-15 by yunhanw-google