project-chip/connectedhomeip #21866: [ESP32][chip-tool] Operational discovery fails using minimalmdns and server interactions

s-mcclain, 2022-08-12:
### Problem
After successfully commissioning the all-clusters-app, chip-tool then attempts to send an OnOff toggle command - but the operational discovery fails every time.

chip-tool log:
```
➜  chip-tool-minimalmdns git:(sve) sudo ./chip-tool onoff toggle 1234 1
Password:
[1660326571236] [81521:5783018] CHIP: [DL] KVS will be written to: file:///tmp/chip_kvs
[1660326571244] [81521:5783018] CHIP: [DL] RequestCallbackOnPendingRead with no dispatch queue: callback may not work (might be ok in tests)
[1660326571248] [81521:5783018] CHIP: [DL] _Init
[1660326571248] [81521:5783018] CHIP: [IN] UDP::Init bind&listen port=0
[1660326571248] [81521:5783018] CHIP: [IN] UDP::Init bound to port=57094
[1660326571248] [81521:5783018] CHIP: [IN] UDP::Init bind&listen port=0
[1660326571248] [81521:5783018] CHIP: [IN] UDP::Init bound to port=63704
[1660326571248] [81521:5783018] CHIP: [IN] BLEBase::Init - setting/overriding transport
[1660326571248] [81521:5783018] CHIP: [IN] TransportMgr initialized
[1660326571248] [81521:5783018] CHIP: [FP] Initializing FabricTable from persistent storage
[1660326571248] [81521:5783018] CHIP: [TS] Last Known Good Time: 2022-08-12T10:31:42
[1660326571248] [81521:5783018] CHIP: [TS] Setting Last Known Good Time to firmware build time 2022-08-12T10:42:29
[1660326571249] [81521:5783018] CHIP: [FP] Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x3CD98ACD03F42C7B, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1660326571250] [81521:5783018] CHIP: [FP] Fabric index 0x2 was retrieved from storage. Compressed FabricId 0xB9EB2EE43EB30E7E, FabricId 0x0000000000000002, NodeId 0x000000000001B669, VendorId 0xFFF1
[1660326571250] [81521:5783018] CHIP: [FP] Fabric index 0x3 was retrieved from storage. Compressed FabricId 0x9DDB5C230CD139F7, FabricId 0x0000000000000003, NodeId 0x000000000001B669, VendorId 0xFFF1
[1660326571252] [81521:5783018] CHIP: [ZCL] Using ZAP configuration...
[1660326571253] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on en5 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:748: Inet Error 0x00000110: Address not found
[1660326571255] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on awdl0 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:748: Inet Error 0x00000110: Address not found
[1660326571256] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on llw0 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:748: Inet Error 0x00000110: Address not found
[1660326571257] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on en3 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:748: Inet Error 0x00000110: Address not found
[1660326571257] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on en3 for address type IPv6: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:788: OS Error 0x02000016: Invalid argument
[1660326571258] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on en1 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:748: Inet Error 0x00000110: Address not found
[1660326571258] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on en1 for address type IPv6: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:788: OS Error 0x02000016: Invalid argument
[1660326571259] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on en2 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:748: Inet Error 0x00000110: Address not found
[1660326571259] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on en2 for address type IPv6: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:788: OS Error 0x02000016: Invalid argument
[1660326571260] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on en4 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:748: Inet Error 0x00000110: Address not found
[1660326571260] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on en4 for address type IPv6: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:788: OS Error 0x02000016: Invalid argument
[1660326571261] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on bridge0 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:748: Inet Error 0x00000110: Address not found
[1660326571261] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on bridge0 for address type IPv6: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:788: OS Error 0x02000016: Invalid argument
[1660326571262] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on utun0 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:748: Inet Error 0x00000110: Address not found
[1660326571263] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on utun1 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:748: Inet Error 0x00000110: Address not found
[1660326571264] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on utun2 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:748: Inet Error 0x00000110: Address not found
[1660326571265] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on utun1000 for address type IPv6: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:788: OS Error 0x02000016: Invalid argument
[1660326571265] [81521:5783018] CHIP: [IN] CASE Server enabling CASE session setups
[1660326571265] [81521:5783018] CHIP: [IN] SecureSession[0x600000860000]: Allocated Type:2 LSID:36936
[1660326571265] [81521:5783018] CHIP: [SC] Allocated SecureSession (0x600000860000) - waiting for Sigma1 msg
[1660326571265] [81521:5783018] CHIP: [DIS] Updating services using commissioning mode 0
[1660326571266] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on en5 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:748: Inet Error 0x00000110: Address not found
[1660326571267] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on awdl0 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:748: Inet Error 0x00000110: Address not found
[1660326571268] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on llw0 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:748: Inet Error 0x00000110: Address not found
[1660326571270] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on en3 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:748: Inet Error 0x00000110: Address not found
[1660326571270] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on en3 for address type IPv6: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:788: OS Error 0x02000016: Invalid argument
[1660326571271] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on en1 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:748: Inet Error 0x00000110: Address not found
[1660326571271] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on en1 for address type IPv6: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:788: OS Error 0x02000016: Invalid argument
[1660326571272] [81521:5783018] CHIP: [DL] MDNS failed to join multicast group on en2 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:748: Inet Error 0x00000110: Address not found
[... cut ...]
scripts/examples/gn_build_example.sh examples/chip-tool chip-tool-minimalmdns-noephemeral chip_mdns=\"minimal\" chip_inet_config_enable_ipv4=true chip_use_ephemeral_port_for_mdns_unicast_query=true
```
- Building with minimalmdns rather than avahi, utilizing ephemeral ports, and enabling server interactions with chip-tool as the Controller
```
// Modifying examples/chip-tool/commands/common/CHIPCommand.cpp
factoryInitParams.enableServerInteractions = true;
```

Using regular ESP32 build with all-clusters-app.

 ### Proposed Solution
Operations should be successful always regardless of using avahi or CHIP's minimalmdns solution.

s-mcclain, 2022-08-12:
And as another note, when `enableServerInteractions` is set to FALSE for chip-tool, operational discovery works just fine and the chip-tool controller is able to send OnOff toggle commands no problem.
[mdns_esp32_nosrv.pcap.zip](https://github.com/project-chip/connectedhomeip/files/9329210/mdns_esp32_nosrv.pcap.zip)

mrjerryjohns, 2022-08-15:
I'm not able to reproduce what you're seeing on my Mac.

Compiling `chip-tool` with `enableServerInteractions` set to false, it's not able to even complete commissioning when I use minimalmdns. Taking out the `chip_mdns` then makes it work for me.

I'm not sure using minimalmdns on Mac is a supported use-case.

@andreilitvin @bzbarsky-apple @woody-apple would you know?

s-mcclain, 2022-08-15:
Through the use of ephemeral ports, I don't see what the issue would be with using the Matter provided minimalmdns solution over Avahi?

bzbarsky-apple, 2022-08-25:
> I'm not sure using minimalmdns on Mac is a supported use-case.

It's not particularly.  It's not tested, and last I tried it some of the broadcast stuff it tries to do by default does not work on Mac.

> I don't see what the issue would be with using the Matter provided minimalmdns solution over Avahi?

What does Avahi have to do with anything?  On Mac what's used when using "platform" mdns is the built-in mdns bits in MacOS, not Avahi.

closed 2022-08-25 by bzbarsky-apple

andy31415, 2022-08-25:
@s-mcclain - don't use minmdns on mac.