project-chip/connectedhomeip #21880: build android chip-tool error in android studio

xufangbo, 2022-08-15:
#### Problem
build android chip-tool in android studio has such error:
~~~
In file included from{chip-tool}/src/crypto/CHIPCryptoPALmbedTLS.cpp:23:
{chip-tool}/src/crypto/CHIPCryptoPAL.h:26:10: fatal error: 'crypto/CryptoBuildConfig.h' file not found
~~~

matterSdkSourceBuild is setting to 'true' in file {chip-tool}/src/android/CHIPTool/gradle.properties

 #### Proposed Solution
-

sunshinehu, 2022-08-17:
I have the same problem when building android chip-tool on M1 Mac

IoT-Project-MDS, 2022-08-22:
Did you solve this problem? i have the same issue.

lliebig, 2022-09-05:
I ran into the same issue and found the missing files in the output folder of the core Matter code.

For example:
`connectedhomeip/out/android_arm64/gen/include`

I have just copied the files to the expected location. But afterwards I ran into another build issue:

`FAILED: connectedhomeip/out/android_arm64/gen/src/controller/java/java.classlist `

lliebig, 2022-09-06:
I started over and realized, that I forgot to execute this in the terminal:

`./scripts/build/build_examples.py --target android-arm64-chip-tool build`

Given in:
https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/android_building.md#building-android-chiptool-from-scripts

felixbaral, 2022-09-15:
i have the same issue, any solution yet?

theyoung, 2022-09-19:
There is a "fatal error: 'crypto/CryptoBuildConfig.h' file not found" error that occurred during the build android artifact. Does any other option exist?

ManBali, 2022-09-26:
ye,i havae the same issue, any one know how to solve it

felixbaral, 2022-09-29:
I am using the app with nordic stuff anyways so i tried their repo and v2.1.0 tag and i did not have the error there: https://github.com/nrfconnect/sdk-connectedhomeip/tree/v2.1.0

fangxiaoying, 2022-09-29:
> Do you know the tag base on which branch? My side didn't encourage the issue before.

fangxiaoying, 2022-09-29:
> I am using the app with nordic stuff anyways so i tried their repo and v2.1.0 tag and i did not have the error there: https://github.com/nrfconnect/sdk-connectedhomeip/tree/v2.1.0

Do you know the tag base on which branch? My side didn't encourage the issue before.

closed 2023-06-10 by stale[bot]