project-chip/connectedhomeip #21881: Restarting the Thread Border Router leads to "No such device" errors

dtodor, 2022-08-15:
Steps to reproduce:
1. Pair a Thread device
2. Interact with device over a CASE session (e.g. OnOff::Toggle)
3. Restart Border Router without restarting the Matter controller
4. Try to interact again and observe the following error:
```
[1660561316.371346][4503:4504] CHIP:CSM: FindOrEstablishSession: PeerId = [1:00000000000003E8]
[1660561316.371377][4503:4504] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1660561316.371403][4503:4504] CHIP:CTL: Found an existing secure session to [1:00000000000003E8]!
[1660561316.371424][4503:4504] CHIP:CTL: OperationalSessionSetup[1:00000000000003E8]: State change 1 --> 5
[1660561316.371466][4503:4504] CHIP:DMG: ICR moving to [AddingComm]
[1660561316.371489][4503:4504] CHIP:DMG: ICR moving to [AddedComma]
[1660561316.371574][4503:4504] CHIP:IN: Prepared secure message 0x2a1d93b4 to 0x00000000000003E8 (1)  of type 0x8 and protocolId (0, 1) on exchange 15705i with MessageCounter:5567535.
[1660561316.371615][4503:4504] CHIP:IN: Sending encrypted msg 0x2a1d93b4 with MessageCounter:5567535 to 0x00000000000003E8 (1) at monotonic time: 00000000002D3671 msec
[1660561316.372120][4503:4509] CHIP:CTL: Failed to handle invoke, error: ../../third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:406: OS Error 0x02000013: No such device
```

If the Matter controller is restarted as well, everything works fine again.

mrjerryjohns, 2022-08-15:
Hmm, this is concerning. If such errors are possible, we need a means to recover from those (which really entails marking the session as defunct like we do automatically when we encounter exchange timeouts) so that a subsequent attempt will correctly right itself.

mrjerryjohns, 2022-08-15:
@jwhui @bukepo is this expected behavior?

mrjerryjohns, 2022-08-15:
Is the border router agent running on the same device that is running the application that is interacting with the Thread device?

jwhui, 2022-08-15:
> @jwhui @bukepo is this expected behavior?

Is the Matter controller binding directly to the `wpan0` interface id? I believe restarting OTBR will recreate the `wpan0` interface.

dtodor, 2022-08-15:
> Is the border router agent running on the same device that is running the application that is interacting with the Thread device?

Yes, the border router and the Matter controller (a daemon in our case) both run on the same device. The `wpan0` will be indeed recreated.

dtodor, 2022-08-15:
> > @jwhui @bukepo is this expected behavior?
> 
> Is the Matter controller binding directly to the `wpan0` interface id? I believe restarting OTBR will recreate the `wpan0` interface.

Not intentionally. But the open sockets might be.

mrjerryjohns, 2022-08-15:
I believe we're passing in an ip address of all 0s, and a `sin6_scope_id` to 0 when calling `bind`. Shouldn't that not result in any issues even if wpan0 restarts?

kghost, 2022-08-16:
our UDP socket implementation are bound to a interface (identified by an interface id, the number before each interface in `ip ad` command), ex `1: ` here.

```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever

```

The peer address is bound a interface when created, there is an [interface id field](https://github.com/project-chip/connectedhomeip/blob/08f09832df5c30ab37267f86c6919f8b23bdfbfb/src/transport/raw/PeerAddress.h#L239) inside peer address object.

```
    Inet::IPAddress mIPAddress   = {};
    Type mTransportType          = Type::kUndefined;
    uint16_t mPort               = CHIP_PORT; ///< Relevant for UDP data sending.
    Inet::InterfaceId mInterface = Inet::InterfaceId::Null();
```

For CASE/PASE server, when a pairing session is staring (receiving Sigma1 or PBKDF req), the peer address is passed from transport layer, containing the interface where the packet is received.

For CASE/PASE client, the peer address is created by mDNS resolver, I'm not sure whether it contains interface id.

To resolve this problem, we can add `persistent` flag to the tun device

```
ioctl(tap_fd, TUNSETPERSIST, 1);
```

On `ot-br` restart, it can reuse the existing tun device, such that the interface id doesn't change when `ot-br` restarts.

dtodor, 2022-08-16:
OK, I will try this out.

dtodor, 2022-08-16:
The interface remains after the OTBR is shut down:
```
wpan0     Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet6 addr: fdb3:126c:11c6:6680:0:ff:fe00:fc10/64 Scope:Global
          inet6 addr: fd54:8b53:eecd:1:4e53:8580:d527:be/64 Scope:Global
          UP POINTOPOINT NOARP MULTICAST  MTU:1280  Metric:1
          RX packets:12 errors:0 dropped:1 overruns:0 frame:0
          TX packets:28 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500 
          RX bytes:2600 (2.5 KiB)  TX bytes:3900 (3.8 KiB)
```

But afterwards it is not possible to start it again:
```
Aug 16 13:19:54 qivicon user.crit otbr-agent[2260]: 49d.18:38:22.031 [C] Platform------: platformConfigureTunDevice() at netif.cpp:1599: Device or resource busy
```

dtodor, 2022-08-16:
Interesting, I can manually delete the `wpan0` interface and then start the OTBR:
```
ip link delete wpan0
```
The Matter controller works afterwards without issues.

dtodor, 2022-08-16:
OK, so partially false alarm. The problem cannot be reproduced on a modern kernel. We seem to have this issue on a 3.4.11 kernel. The `persistent` flag as suggested above leads to the `Device or resource busy` error. Further, even after manually deleting the `wpan0`, we still get `Device or resource busy` when starting the OTBR.

dtodor, 2022-08-16:
I've managed to get the persistent `wpan0` interface working on the older kernel. Now, the Matter controller works properly even after restarting the OTBR. Thanks everyone for the help!

closed 2022-08-16 by dtodor

kghost, 2022-08-16:
@dtodor If you reuse the tun device by calling `TUNSETIFF` with `IFF_TUN_EXCL` flag (which is the default behavior), you will get EBUSY error. The correct way is remove EXCL flag in this case.