project-chip/connectedhomeip #21892: chip-tool accesscontrol write acl accepts any random arguments

bluebin14, 2022-08-15:
#### Problem
chip-tool accesscontrol write acl accepts any random arguments, e.g.

`./chip-tool accesscontrol write acl '[{"water":123,"privilege":5,"authMode":2,"subjects":[],"targets":null}]' 12344321 0`

 #### Proposed Solution
Do not accept random arguments.

bluebin14, 2022-08-15:
This is true for all JSON parsing, not just acesscontrol.

bzbarsky-apple, 2022-08-25:
@bluebin14 So just to make sure I understand, the issue there is the `"water": 123` not leading to an error?

bluebin14, 2022-08-25:
@bzbarsky-apple yes I was asked to file an issue so someone can determine if it's an issue or not. It came up when incorrect fabricId was being passed which are now ignored anyway.

bzbarsky-apple, 2022-08-25:
@bluebin14 `fabricId`, or `fabricIndex`?  (I suspect the fix for fabricIndex would be a bit different from the fix for general "not a known member"....)

bluebin14, 2022-08-26:
fabricIndex

bzbarsky-apple, 2022-08-26:
Ok.  The fabricIndex bit is in fact allowed to be sent by the spec, so excluding it would be a bit odd.

closed 2023-02-24 by andy31415