project-chip/connectedhomeip #21894: Make sure we sending out a `CLOSE_SESSION` when expiring sessions

tehampson, 2022-08-15:
#### Problem
* Issue was brought up https://github.com/project-chip/connectedhomeip/pull/21886#discussion_r945822956

 #### Proposed Solution
* Peform audit and add call to send CLOSE_SESSION` when session is marked as expired.

tehampson, 2022-08-16:
Okay did an audit today and it doesn't look like this is implemented on session eviction. I think all that is needed is that we call something similar to `CloseSessionCommand::CloseSession` from `examples/chip-tool/commands/pairing/CloseSessionCommand.cpp` inside something like `SecureSession::MarkForEviction()`

mrjerryjohns, 2022-08-17:
This is not a super high priority for v1.0. The consequence of not doing so is that we may have more 'mis-matched' sessions between a set of peers, but I think until we see real empirical data from real-world data, it's not clear this is really a problem.

msandstedt, 2022-09-12:
> The consequence of not doing so is that we may have more 'mis-matched' sessions between a set of peers, but I think until we see real empirical data from real-world data, it's not clear this is really a problem.

The additional consequence is the need to wait for at least MRP timeout to detect session disappearance in the peer, instead of a possible happy case where this is known immediately.

Said another way, CLOSE_SESSION is an optimization.

closed 2023-05-04 by stale[bot]