project-chip/connectedhomeip #21895: M5 Matter accessory crashed

shah-kartik, 2022-08-15:
CHIP SHA:
aca4b94f9220f2dd23c44da5dbfcdf8f212f5e4d

Traceback (most recent call last):
  File "/Users/apple/tools/esp-idf/tools/idf_monitor.py", line 351, in <module>
    main()
  File "/Users/apple/tools/esp-idf/tools/idf_monitor.py", line 342, in main
    monitor.main_loop()
  File "/Users/apple/tools/esp-idf/tools/idf_monitor.py", line 155, in main_loop
    self._main_loop()
  File "/Users/apple/tools/esp-idf/tools/idf_monitor.py", line 255, in _main_loop
    super()._main_loop()
  File "/Users/apple/tools/esp-idf/tools/idf_monitor.py", line 200, in _main_loop
    self.serial_handler.handle_serial_input(data, self.console_parser, self.coredump,
  File "/Users/apple/tools/esp-idf/tools/idf_monitor_base/serial_handler.py", line 94, in handle_serial_input
    self.logger.handle_possible_pc_address_in_line(line)
  File "/Users/apple/tools/esp-idf/tools/idf_monitor_base/logger.py", line 132, in handle_possible_pc_address_in_line
    translation = lookup_pc_address(m.group(), self.toolchain_prefix, self.elf_file)
  File "/Users/apple/tools/esp-idf/tools/idf_monitor_base/output_helpers.py", line 46, in lookup_pc_address
    translation = subprocess.check_output(cmd, cwd='.')
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/subprocess.py", line 411, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['xtensa-esp32-elf-addr2line', '-pfiaC', '-e', '/Users/apple/connectedhomeip/examples/all-clusters-app/esp32/build/chip-all-clusters-app.elf', '0x4353E0B3']' returned non-zero exit status 1.
idf_monitor failed with exit code 1

full logs are attached here
[Matter_M5_Crash.pdf](https://github.com/project-chip/connectedhomeip/files/9340046/Matter_M5_Crash.pdf)

bzbarsky-apple, 2022-08-25:
@dhrishi this looks like the esp32 Python scripts hitting exceptions, not the m5stack itself crashing....

dhrishi, 2022-08-26:
@bzbarsky-apple Will check

dhrishi, 2022-08-27:
@shah-kartik Now we use IDF v4.4.2 with Matter and not v4.4.1. Can you please try with the v4.4.2 IDF and see if you still face this issue? Meanwhile, I will check with the tools team on this.

dobairoland, 2022-09-05:
Hi @shah-kartik. Could you please try to run the following command and let us know about the received output? Please run it in the same environment (terminal) where you are receiving the above issue.

```sh
xtensa-esp32-elf-addr2line -pfiaC -e /Users/apple/connectedhomeip/examples/all-clusters-app/esp32/build/chip-all-clusters-app.elf 0x4353E0B3
```

I suspect that some library is missing from your computer. The output of the command will help us to determine....

shah-kartik, 2022-09-28:
Hi, I have not seen this recently. Feel free to close it. I will re-open if it shows up again and provide the output to the requested command as well.

dhrishi, 2022-09-30:
Sure, thanks!

closed 2022-09-30 by dhrishi