project-chip/connectedhomeip #21896: Out of curiosity, in which scenarios is it useful to loop over all sessions instead of just do `holder->MarkAsDefunct`? Do clients try establish another session to a node without marking the previous one as defunct?

woody-apple, 2022-08-15:
Out of curiosity, in which scenarios is it useful to loop over all sessions instead of just do `holder->MarkAsDefunct`? Do clients try establish another session to a node without marking the previous one as defunct?

_Originally posted by @Damian-Nordic in https://github.com/project-chip/connectedhomeip/pull/21841#discussion_r944566191_

mrjerryjohns, 2022-08-17:
It's possible for the server to establish a CASE session to the client too, which can result in multiple such sessions between the two. In practice, this should be quite in-frequent, but in theory, it is possible.

It's also possible for a session that was previously marked defunct to be re-instated as active (if we get traffic back from the peer). That can also result in more than 1 session to the peer.

Damian-Nordic, 2022-08-18:
Ok, that answers my question. Thanks @mrjerryjohns!

closed 2022-08-18 by Damian-Nordic