project-chip/connectedhomeip #21898: ZAP external attributes are size 0, and fails datalen check when trying to write

ryetree1, 2022-08-16:
When you mark an attribute as external (storing in external flash), the size field in the generated endpoints.h gets set to 0 for that attribute.   

When that attribute is written to it goes through WriteSingleClusterData() in ember-compatibility-functions.cpp.   In that function is does a check "if (dataLen > attributeMetadata->size)", which fails since the attribute size is 0.  This results in an error 0x87, constraint error.

mrjerryjohns, 2022-08-17:
The above statements don't quite line up what I see in the code. External attributes do still have valid size fields setup in the generated AttributeMetadata tables:

```
            { 0x00000001, ZAP_TYPE(INT16U), 2, ZAP_ATTRIBUTE_MASK(EXTERNAL_STORAGE), ZAP_EMPTY_DEFAULT() }, /* RebootCount */      \
            { 0x00000002, ZAP_TYPE(INT64U), 8, ZAP_ATTRIBUTE_MASK(EXTERNAL_STORAGE), ZAP_EMPTY_DEFAULT() }, /* UpTime */           \
```

The only ones that don't have a valid size are those of type struct or array. Those cannot be handled by the Ember layer anyways.

ryetree1, 2022-08-17:
strange, all my external attributes have a size 0.  My ram attributes have actual size

(example):

/* Endpoint: 1, Cluster: On/Off (server) */ \
  { 0x00000000, ZAP_TYPE(BOOLEAN), 0, ZAP_ATTRIBUTE_MASK(EXTERNAL_STORAGE), ZAP_EMPTY_DEFAULT() }, /* OnOff */  \
  { 0x00004000, ZAP_TYPE(BOOLEAN), 0, ZAP_ATTRIBUTE_MASK(EXTERNAL_STORAGE), ZAP_EMPTY_DEFAULT() }, /* GlobalSceneControl */  \
  { 0x00004001, ZAP_TYPE(INT16U), 0, ZAP_ATTRIBUTE_MASK(EXTERNAL_STORAGE) | ZAP_ATTRIBUTE_MASK(WRITABLE), ZAP_EMPTY_DEFAULT() }, /* OnTime */  \
  { 0x00004002, ZAP_TYPE(INT16U), 0, ZAP_ATTRIBUTE_MASK(EXTERNAL_STORAGE) | ZAP_ATTRIBUTE_MASK(WRITABLE), ZAP_EMPTY_DEFAULT() }, /* OffWaitTime */  \

Maybe its an issue in my zap revision, will double check it all.

bzbarsky-apple, 2022-08-25:
@ryetree1 Which exact SHA of Matter and ZAP are you using?  The behavior here changed in #19812 and it's possible you are using something from before then.

ryetree1, 2022-08-29:
I updated the ZAP and this issue seems to be addressed.  Closing issue.

closed 2022-08-29 by ryetree1