project-chip/connectedhomeip #21912: Fails to store Thread operational credentials in KVS on platforms that does not support 0 length writes

ReneJosefsen, 2022-08-16:
#### Problem
When sending the Thread operational dataset during pairing, an unknown error is returned if the platform does not support KVS writes with length set to 0.

The changes introduced in https://github.com/project-chip/connectedhomeip/commit/ce654029a4fae7e46df049b5b48a0cd8b67cbfbf removed the handling of 0 length writes and seems to expect that KVS rejects this nicely rather then returning an error. (Specifically the change on line 213 is problematic for some platforms)

 #### Proposed Solution
Re-introduce the empty write if the mStagingNetwork is empty or make sure all platforms nicely rejects KVS writes of 0 length.

ReneJosefsen, 2022-08-16:
Sure, if that is the agreement then that is the place to fix it. What is the agreed handling of 0-length writes, just return CHIP_NO_ERROR?

Damian-Nordic, 2022-08-16:
Yes, but the subsequent read of such a value should also return CHIP_NO_ERROR despite not returning any bytes, so you must store something on such a write :). See `PersistentStorageAudit.cpp`, the line commented with "Attempt an empty key write with either nullptr or zero size works".

ReneJosefsen, 2022-08-16:
How many platforms has this been tested on?

Looking a bit around in the different KeyValueStoreManagerImpl files I do see a few implementations which looks like it would fail the test you are referencing:

Darwin:
```
CHIP_ERROR KeyValueStoreManagerImpl::_Put(const char * key, const void * value, size_t value_size)
{
    ReturnErrorCodeIf(key == nullptr, CHIP_ERROR_INVALID_ARGUMENT);
```

EFR32:
```
CHIP_ERROR KeyValueStoreManagerImpl::_Put(const char * key, const void * value, size_t value_size)
{
    VerifyOrReturnError(key != nullptr, CHIP_ERROR_INVALID_ARGUMENT);
```

CC13x2_26x2:
```
CHIP_ERROR KeyValueStoreManagerImpl::_Put(const char * key, const void * value, size_t value_size)
{
    VerifyOrReturnError(key, CHIP_ERROR_INVALID_ARGUMENT);
    VerifyOrReturnError(value, CHIP_ERROR_INVALID_ARGUMENT);
```

ESP32:
```
CHIP_ERROR KeyValueStoreManagerImpl::_Put(const char * key, const void * value, size_t value_size)
{
    VerifyOrReturnError(value, CHIP_ERROR_INVALID_ARGUMENT);
```

Damian-Nordic, 2022-08-16:
You can see platforms whose maintainers claim to pass the audit in https://github.com/orgs/project-chip/projects/3. 

Darwin and EFR32 just validate that the key is not null, but value can be null, right? ESP32 indeed seems to require that value be not null, but the size isn't checked. The audit on ESP32 probably passes because `KvsPersistentStorageDelegate` makes sure that the value passed to `KeyValueStoreManager` is not null (although the size can be 0). Thus if I'm not mistaken, only CC13x2_26x2 doesn't correctly handle zero-length values out of those listed by you.

ReneJosefsen, 2022-08-16:
I do not have access to the link.

But you are completely correct, Darwin and EFR32 checks the key and not the value so they are fine, my mistake. ESP32 still seems a bit tricky to me, since it would return CHIP_ERROR_INVALID_ARGUMENT on the storage.SyncSetKeyValue with nullptr but the test expects CHIP_NO_ERROR, but if I understand you correctly there is an additional layer which handles this or?

CC13x2_26x2 indeed have issues with the 0-length. I have not checked all platforms, but If this is only a CC13x2_26x2 issue, I can rename the title of the issue to indicate this.

srickardti, 2022-08-16:
The underlying storage mechanism for NVS on the CC13X2_26X2 can't handle an element of zero length. It doesn't really make sense to set an empty value.

This reminds me of: https://wiki.c2.com/?NullConsideredHarmful

We can implement this by checking if a key exists without a value assume it is a zero length element in that case.

tcarmelveilleux, 2022-08-17:
> It doesn't really make sense to set an empty value.

While this may be your opinion, we rely on this in several places and I have a lot of products here that allow empty values.

Consider the humble Python dict:

```
{
   "a": ""
}
```

This is definitely different than
```
{ } 
```

Damian-Nordic, 2022-08-17:
@srickardti We also struggled with that too as Zephyr's settings API also doesn't support empty values. In Zephyr's implementation of the KVS we worked around this by putting magic bytes into the store as the representation of the empty value. You can probably do something similar. It is not ideal, but good enough solution.

srickardti, 2022-08-17:
> > It doesn't really make sense to set an empty value.
> 
> While this may be one opinion, we rely on this in several places and I have a lot of products here that allow empty values.

Yeah, the difference is really just in the assumptions each system makes. The NVS system in our SDK doesn't have a default value option so deleting an item is logically equivalent to setting it to empty. Having a `NULL` in the storage system for the ConfigurationManager should mean use the default value. So I understand that we need to be able to set a `""` value.

ReneJosefsen, 2022-08-17:
Or at minimum fake it by using some magic value as @Damian-Nordic is talking about?

tehampson, 2022-08-17:
Sorry for coming in late here... For some reason I am not getting emails/notification when I am mentioned. I will work on fixing that.

Thanks Damian for making all the points I would have made. `PersistentStorageDelegate::SyncSetKeyValue` was defined to allow setting values of 0 size (and when value is nullptr). This decision was made before I started working on matter, but I am guessing this decision was made earlier on so that application relying on this could depend on this being available. Just like [tcarmelveilleux](https://github.com/tcarmelveilleux) mentioned there are cases where this is important. This has been in the API contract of `PersistentStorageDelegate` for quite some time.

Overall my two cents is this issue should be renamed to have CC13X2_26X2 pass the PersistentStorageDelegate audit:

To run the audit on your device you simply need to add the gn build arg `chip_support_enable_storage_api_audit=true` to your build.

To build for Linux, for example:
In SDK root: `scripts/examples/gn_build_example.sh examples/all-clusters-app/linux out/debug/standalone chip_config_network_layer_ble=false chip_support_enable_storage_api_audit=true`
Then `./out/debug/standalone/chip-all-clusters-app` and the audit will run in early init. You can observe it in the logs.

The output of the Audit is a log:
* ==== PersistentStorageDelegate API audit: SUCCESS ====
* ==== PersistentStorageDelegate API audit: FAILED: %d/%d failed assertions ====

srickardti, 2022-08-17:
@tehampson In the past the persistent-storage example application was the audit for the KVS system. Has that changed? If so, should we just remove that example to avoid confusion?

tehampson, 2022-08-17:
I think I see the issue. We actually never fixing discrepancies between `PersistentStorageDelegate` and `KeyValueStoreManager`. It seems like we made it very clear about `PersistentStorageDelegate` but never made that clarification in `KeyValueStoreManager`.

The audit is more for `PersistentStorageDelegate` and that example uses `KeyValueStoreManager`. Overall the reason we did the audit the way we did is that it allowed all platforms to quickly test to make sure they are compliant. There is no good way for platforms to use a common device test that runs across platforms. I don't have enough background to make this call, but I would say the test in `examples/persistent-storage/` is not as robust as the audit.

I think that maybe we should make sure to make `KeyValueStoreManager` APIs inline with `PersistentStorageDelegate` at this point.

tcarmelveilleux, 2022-08-17:
`KeyValueStoreManager` is platform level. You could implement a platform without ever using it, but most examples show it.

`PersistentStorageDelegate` is a storage abstraction used by all common code above the platform layer. That one is the only one whose API contract we "uniformly" try to validate.

tehampson, 2022-08-17:
Based on what tcarmelveilleux@ is saying I think we cannot make the same assumptions about `KeyValueStoreManager` based on APIs and audit of `PersistentStorageDelegate`. So while CC13X2_26X2 does need to pass the `PersistentStorageDelegate` audit and that should be a github issue on it's own, it doesn't mean that `KeyValueStoreManager` has the same API contract :/ 

For that reason maybe `BackupConfiguration()` should handle the 0 size case differently. Sorry for the back and forth here.

PR #21977 (2022-08-23): Update KVS to handle zero length elements
#### Problem

fixes #21912

Fix the CC13X2_26X2 KVS implementation to allow for zero length values.

#### Change overview

If a key item exists without a value item, we now assume that the value is a zero length element. The read/write/delete functions have been updated to reflect this new assumption. The KVS interfaces have been updated to not reject values of zero length.

The persistent-storage example has been updated with a new test for binary values of zero length.

#### Testing

the persistent-storage application was run and produced passing res [... cut]
closed 2022-08-23 by andy31415