project-chip/connectedhomeip #21917: IOS16 device to configure network for esp32, CASE session establishment fails

66sheng, 2022-08-16:
**Problem**
When using IOS16 device to configure network for esp32, CASE session establishment fails
If you use chip-tool, you can configure the network normally

Test Equipment:
iphone11  ios16.0(20A5328h)
iphone TV tvos16.0(20J5344f)
esp32

esp32 code:
sve：aa9457e6b94b735076dff6297176183bf9780177 or TE9： cfc35951be66a664a6efdadea56d1b8ea6e63e96
Test with all-cluster-app，same test result

Test steps:
Use the iphone home app to scan the QR code to configure the esp32
Can't pass the CASE session after connecting to wifi
e.g. esp32log ： fail_esp32_ios16_1
When the esp32 establishes a CASE session with the iphone, the sigma3 data packet is not received, resulting in the failure of the network configuration

esp32 log：
I (61424) chip[IN]: CASE Server received Sigma1 message. Starting handshake. EC 0x3ffc70e8
I (61434) chip[IN]: CASE Server disabling CASE session setups
I (61444) chip[IN]: Prepared unauthenticated message 0x3ffe6424 to 0x0000000000000000 (0)  of type 0x10 and protocolId (0, 0) on exchange 34435r with MessageCounter:55905246.
I (61454) chip[IN]: Sending unauthenticated msg 0x3ffe6424 with MessageCounter:55905246 to 0x0000000000000000 at monotonic time: 000000000000ECD5 msec
I (61484) chip[SC]: Received Sigma1 msg
I (61484) chip[SC]: CASE matched destination ID: fabricIndex 1, NodeID 0x00000000AF7B9C6E
I (62454) chip[IN]: Prepared unauthenticated message 0x3ffc739c to 0x0000000000000000 (0)  of type 0x31 and protocolId (0, 0) on exchange 34435r with MessageCounter:55905247.
I (62474) chip[IN]: Sending unauthenticated msg 0x3ffc739c with MessageCounter:55905247 to 0x0000000000000000 at monotonic time: 000000000000F0CB msec
I (62484) chip[SC]: Sent Sigma2 msg
E (62484) chip[DL]: Long dispatch time: 1069 ms, for event type 3
I (62464) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (62794) chip[IN]: Sending unauthenticated msg 0x3ffc739c with MessageCounter:55905247 to 0x0000000000000000 at monotonic time: 000000000000F211 msec
I (63124) chip[IN]: Sending unauthenticated msg 0x3ffc739c with MessageCounter:55905247 to 0x0000000000000000 at monotonic time: 000000000000F35D msec
I (63624) chip[IN]: Sending unauthenticated msg 0x3ffc739c with MessageCounter:55905247 to 0x0000000000000000 at monotonic time: 000000000000F550 msec
I (63644) chip[SWU]: Stopping the watchdog timer
I (63644) chip[SWU]: Starting the periodic query timer, timeout: 86400 seconds
I (63644) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x002a', EndPoint ID: '0x00', Attribute ID: '0x0002'
I (63654) app-devicecallbacks: Unhandled cluster ID: 42
I (63674) app-devicecallbacks: Current free heap: 76172

I (63674) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x002a', EndPoint ID: '0x00', Attribute ID: '0x0003'
I (63684) app-devicecallbacks: Unhandled cluster ID: 42
I (63684) app-devicecallbacks: Current free heap: 76860

I (64444) chip[IN]: Sending unauthenticated msg 0x3ffc739c with MessageCounter:55905247 to 0x0000000000000000 at monotonic time: 000000000000F887 msec
I (64494) chip[EM]: Received message of type 0x40 with protocolId (0, 0) and MessageCounter:262016980 on exchange 34435r
E (64494) chip[SC]: Received error (protocol code 2) during pairing process: 54
E (64504) chip[IN]: CASE Session establishment failed: 54
I (64504) chip[IN]: Prepared unauthenticated message 0x3ffe63e4 to 0x0000000000000000 (0)  of type 0x10 and protocolId (0, 0) on exchange 34435r with MessageCounter:55905248.

e.g. esp32log ：fail_esp32_ios16_2
After the esp32 is connected to the wifi, it does not receive the sigma1 data packet, causing the network configuration to fail

[fail_esp32_ios16_1.txt](https://github.com/project-chip/connectedhomeip/files/9350026/fail_esp32_ios16_1.txt)
[fail_esp32_ios16_2.txt](https://github.com/project-chip/connectedhomeip/files/9350027/fail_esp32_ios16_2.txt)

mrjerryjohns, 2022-08-17:
FYI @vivien-apple @woody-apple

woody-apple, 2022-09-06:
Darwin Issue Review: We believe this is resolved. Can you try this on a more recent iOS? If this is still happening, please let us know with newer logs!

closed 2022-09-06 by woody-apple

ti-tpan, 2022-09-09:
Hi @woody-apple ,

I am seeing similar issues.

**Setup**:

- Apple TV 4K, tvOS 16.0 (20J373)
- iPad 9th gen, iPadOS 16.1 (20B5027f)
- connectedhomeip commit: cfc35951be66a664a6efdadea56d1b8ea6e63e96 (TE9)
- Matter Thread/BLE device: CC2652R7

It seems that the communication from Matter-device to iPad is not getting through somehow.

[cc2652r7_log.txt](https://github.com/project-chip/connectedhomeip/files/9531489/cc2652r7_log.txt)

For example, in the attached log, the Sigma1 is sent multiple times to the Matter-device (see multiple acks for MessageCounter:56681355).
Also, the Sigma2 does not seem to arrive at the iPad (see re-transmissions of MessageCounter:1345719256).

Appreciate any help!

Thanks,
Toby