project-chip/connectedhomeip #21930: Chip-device-ctrl fails commissioning at step 18

jmartinez-silabs, 2022-08-16:
#### Problem
while trying to run python src/python_testing/TC_SC_3_6.py, I encounter this error at commissioning. It uses the python chip-device-ctrl and it fails at commissioning step 18. The commissioning doesn’t fail using chip-tool. 
 
```
[1660666710.492411][12899:12904] CHIP:CTL: Successfully finished commissioning step 'ThreadNetworkEnable'
[1660666710.492461][12899:12904] CHIP:CTL: Commissioning stage next step: 'ThreadNetworkEnable' -> 'FindOperational'
[1660666710.492529][12899:12904] CHIP:CTL: Performing next commissioning step 'FindOperational'
[1660666710.501951][12899:12904] CHIP:DL: Avahi re-register required
[1660666710.503697][12899:12904] CHIP:CTL: Device connection failed. Error ../../src/lib/dnssd/Discovery_ImplPlatform.cpp:626: CHIP Error 0x00000003: Incorrect state
[1660666710.503772][12899:12904] CHIP:CTL: Error on commissioning step 'FindOperational': '../../src/lib/dnssd/Discovery_ImplPlatform.cpp:626: CHIP Error 0x00000003: Incorrect state'
[1660666710.503809][12899:12904] CHIP:CTL: Failed to perform commissioning step 18
[1660666710.503847][12899:12904] CHIP:CTL: Going from commissioning step 'FindOperational' with lastErr = '../../src/lib/dnssd/Discovery_ImplPlatform.cpp:626: CHIP Error 0x00000003: Incorrect state' -> 'Cleanup'
[1660666710.503909][12899:12904] CHIP:CTL: Performing next commissioning step 'Cleanup' with completion status = '../../src/lib/dnssd/Discovery_ImplPlatform.cpp:626: CHIP Error 0x00000003: Incorrect state'
[1660666710.503948][12899:12904] CHIP:CTL: Successfully finished commissioning step 'Cleanup'
Failed to commission: 3
[MatterTest] 08-16 16:18:30.511 INFO [Test] test_run_commissioning FAIL
[MatterTest] 08-16 16:18:30.519 INFO Summary for test class CommissionDeviceTest: Error 0, Executed 1, Failed 1, Passed 0, Requested 1, Skipped 0
[MatterTest] 08-16 16:18:30.520 WARNING Abort all subsequent test classes. Reason: Details=All remaining tests aborted due to: Failed to commission node, Extras=None
[MatterTest] 08-16 16:18:30.522 INFO Summary for test run MatterTest@08-16-2022_16-18-23-609:
Total time elapsed 6.911022891999892s
Artifacts are saved in "/tmp/matter_testing/logs/MatterTest/08-16-2022_16-18-23-609"
Test results: Error 0, Executed 1, Failed 1, Passed 0, Requested 1, Skipped 0
```

commit 5a052078c8a82cb062974130527b477ca71582e7
Chip-device-ctrl  built with `./scripts/build_python.sh -m platform -i separate`

command run for the test
`rm -f admin_storage.json && python src/python_testing/TC_SC_3_6.py -m ble-thread --discriminator 3840 -p 20202021 --thread-dataset-hex 0e080000000000000000000300000b35060004001fffe00208dead00beef00cafe0708fddead00beef000005101b145f341deec678b3f3b62c88757f29030a4f70656e546872656164010209ae04104b85d8385249e10d22c060bac5f89de50c0402a0f7f8 --ble-interface 0`

Log:
[chip-device-ctrl_failure.txt](https://github.com/project-chip/connectedhomeip/files/9352840/chip-device-ctrl_failure.txt)

tcarmelveilleux, 2022-08-16:
It works for me on LInux. It totally looks like an Anahi issue

tehampson, 2022-08-16:
While it looks like an Avahi issue, I am wondering is maybe the change to what is done in FindOperational

I am wondering if you are having this issue because of the change over here:
https://github.com/project-chip/connectedhomeip/pull/21886

Can you try reverting that commit and seeing if things start working for you

jmartinez-silabs, 2022-08-17:
Closing the issue. could not reproduce on another PI. I reinstalled Avahi on the problematic PI it fixed the issue.

closed 2022-08-17 by jmartinez-silabs