project-chip/connectedhomeip #21967: Doxygen build...

achsoftathome, 2022-08-17:
Hi, I made some modifications to allow the construction of the documentation ( doxygen) and I replace the logo by the one of Matter. The problem is induced by the symbolic links which are everywhere. They point to the root of the project, the tool enters in recursion...crash...
Could I make a pull request or pass these two files to someone in the project, tell me?
Arnaud

bzbarsky-apple, 2022-08-25:
@achsoftathome, you can absolutely make a pull request!

andy31415, 2022-09-06:
Also consider changes in https://github.com/project-chip/connectedhomeip/pull/19950

Those were to large to allow for 1.0, however it shows a large update to our doc layout.