project-chip/connectedhomeip #21989: Test_TC_CC_3_1.yaml test has not fully correct step.

AntonGrey, 2022-08-18:
#### Test case implementation has wrong 3c step. According to the test plan on this step CurrentHue value should be 0, but in the test value checks with bounds from 0 to 15. This can lead false failing the test.
 * expected behavior: expected values should be in the middle of accepted values range. For this step accepted range should be from 240 to 255 and from 0 to 15.
 * actual behavior: accepted range is from 0 to 15
 * reproduce steps: 
1.  `./chip-tool colorcontrol move-to-hue 60 0 0 0 1 1`
2. `./chip-tool colorcontrol move-to-hue 135 1 300 0 1 1`
3. After 10 seconds `./chip-tool colorcontrol read current-hue 1 1`

 #### Proposed Solution
Extend the accepted values range so that the expected value is in the middle of the accepted range.

raju-apple, 2022-08-21:
@AntonGrey in future can you please file the test scripts issues [here](https://github.com/CHIP-Specifications/chip-test-scripts/issues) and cert tool issues [here](https://github.com/CHIP-Specifications/chip-certification-tool/issues) please ?

ifyall, 2022-08-23:
This was refiled in the scripts project and can be closed.

closed 2022-08-23 by cjandhyala