project-chip/connectedhomeip #21998: [TC-LVL-2.1] OnLevel value is null instead of integer

adamb-q, 2022-08-18:
Step 9 in TC-LVL-2.1 returns null value as OnLevel, while test plan says that is must be larger than MinLevel which is 1. Tested on 1bc660e70fd233dd0ee42b147e428ce8f8a3338d usin x64 chip-tool and platform qpg.

[TC_LVL_2_1.1660821562.log](https://github.com/project-chip/connectedhomeip/files/9373641/TC_LVL_2_1.1660821562.log)
[TC_LVL_2_1.1660821562_DUT.log](https://github.com/project-chip/connectedhomeip/files/9373642/TC_LVL_2_1.1660821562_DUT.log)

bzbarsky-apple, 2022-08-25:
`null` is a perfectly valid value for OnLevel.

Filed https://github.com/CHIP-Specifications/chip-test-plans/issues/2105

franck-apple, 2022-09-09:
closing, as the test plan issue was created.

closed 2022-09-09 by franck-apple