project-chip/connectedhomeip #21999: [TC-CADMIN-1.15] Read attribute failure even though expected outcome matches

manjunath-grl, 2022-08-18:
Summary: 
While updating script TC-CADMIN-1.15 in step 8 we came across an issue in which the expected response should be the FAILURE and YAML step also expecting the FAILURE response but when executed the following steps fails and terminates execution even though expected outcome matches.
Diff link:
https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:Attribute_read_Error
Yaml screenshot:
[image]
Execution logs:
[CADMIN_1_15.zip](https://github.com/project-chip/connectedhomeip/files/9373583/CADMIN_1_15.zip)
Testplan reference:
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/multiplefabrics.adoc#tc-cadmin-1-15-removing-fabrics-from-dut-and-fabric-index-enumeration-using-ecm-dut-commissionee
[image]

manjunath-grl, 2022-08-23:
Please find the YAML file below:
[Test_TC_CADMIN_1_15.zip](https://github.com/project-chip/connectedhomeip/files/9403889/Test_TC_CADMIN_1_15.zip)

bzbarsky-apple, 2022-08-26:
The "FactoryReset" command at the beginning of that YAML is what's breaking things.  It should not be there @manjunath-grl

bzbarsky-apple, 2022-08-26:
Ah, actually if I take out that FactoryReset things are still failing.  The reason for that is that we get a _client side_ error on the read attempt because there is no session, which is not turned into any sort of FAILURE-like thing in terms of the YAML, because the attempt to _start_ a read fails synchronously:
```
2022-08-25 23:03:27.441 ERROR   23:03:27.147 - TEST OUT  : [1661483007146] [90794:12086340] CHIP: [TOO]  ***** Test Step 11 : TH_CR2 reads the Basic Information Clusters NodeLabel mandatory attribute of DUT_CE
2022-08-25 23:03:27.441 ERROR   23:03:27.147 - TEST OUT  : 
2022-08-25 23:03:27.442 ERROR   23:03:27.147 - TEST OUT  : [1661483007147] [90794:12086340] CHIP: [TOO] 	cluster 0x0000_0028, attribute: 0x0000_0005, endpoint 0
2022-08-25 23:03:27.442 ERROR   23:03:27.147 - TEST OUT  : [1661483007147] [90794:12086340] CHIP: [TOO] Sending ReadAttribute to:
2022-08-25 23:03:27.442 ERROR   23:03:27.147 - TEST OUT  : [1661483007147] [90794:12086340] CHIP: [DMG] SendReadRequest ReadClient[0x7f8de3904080]: Sending Read Request
2022-08-25 23:03:27.442 ERROR   23:03:27.148 - TEST OUT  : [1661483007147] [90794:12086340] CHIP: [TOO]  ***** Test Failure: ../../../examples/chip-tool/third_party/connectedhomeip/src/app/ReadClient.cpp:309: CHIP Error 0x00000077: Missing secure session
```
@vivien-apple

andy31415, 2022-08-31:
Is this a cert blocker? we seem to say "we expect an error" and we actually do get an error, but not something that yaml can interpret.

if so, should the temporary fix be to make this test manual?

bzbarsky-apple, 2022-08-31:
Making this test manual is an option, yes.

andy31415, 2022-09-01:
@cjandhyala - lets update this test to manual if that is the quickest path.

I would accept other patches in 1.0 as well, but this cannot block 1.0 as the behaviour seems fine and we just have a deficiency in test runners.

raju-apple, 2023-05-15:
@vivien-apple