project-chip/connectedhomeip #22001: Emits events for commissioning window open/close and commissioning session start/stop

shubhamdp, 2022-08-18:
#### Problem
Right now, `CommissioningSessionStarted`, `CommissioningSessionStopped`, `CommissioningWindowOpened`, and `CommissioningWindowClosed` are provisioned using [class AppDelegate](https://github.com/project-chip/connectedhomeip/blob/master/src/app/server/AppDelegate.h).

And `CommissioningComplete` and `FailSafeTimerExpired` are provisioned using the chip device events.

If we convert these callback APIs in AppDelegate class then these events can be handled asynchronously.
All the commissioning related events will be delivered by the same method.
Applications won't have to implement the AppDelegate and pass its to Server.

 #### Proposed Solution
Change callback APIs in AppDelegate to events

tcarmelveilleux, 2022-08-18:
AppDelegate is better because it doesn't force using Matter event loop to process this, and can actually defer to application logic. This also simplifies writing tests. We should probably rather add more methods to AppDelegate

shubhamdp, 2022-08-22:
AppDelegate callbacks are also called in Matter thread context. So, to me both seems similar with difference being callback will be called immediately and events will be bit delayed. Also, to defer the application logic both callbacks and events can delegate the heavy work to some other thread.

Agree that AppDelegate does simplifies writing tests.

shubhamdp, 2022-08-25:
Closing this and opened #22165

closed 2022-08-25 by shubhamdp