project-chip/connectedhomeip #22035: [Yaml] The CADMIN yaml test factory reset prompt

doublemis1, 2022-08-19:
#### Problem

The CADMIN yaml scripts (e.g. https://github.com/project-chip/connectedhomeip/blob/master/src/app/tests/suites/certification/Test_TC_CADMIN_1_3.yaml) inform us to proceed the factory reset, while the next step our DUT need to be already in the Matter Network
```
 - label: "TH_CR1 starts a commissioning process with DUT_CE"
      cluster: "DelayCommands"
      command: "WaitForCommissionee"
      PICS: CADMIN.S
      arguments:
          values:
              - name: "nodeId"
                value: nodeId
```
The Factory reset means that the device is back to the original state (not commissioned).
The same issue can be also observed in the test Harness tests while we want to factory reset the device and then the test script expects that the device is in the matter network.

 #### Proposed Solution

Add possibilities to provide a network data (e.g. thread dataset), qr-code/manual pairing code for the first commissioning.
Add a step to perform commissioning right after factory reset (if necessary)

mrjerryjohns, 2022-08-23:
@manjunath-grl, could you help take a look?

bzbarsky-apple, 2022-08-26:
Yeah, "WaitForCommissionee" is not "starts a commissioning process with DUT_CE".  

But also, the "reset to factory defaults" _always_ holds for YAML tests, no?  So that manual reset should not be there at all.

doublemis1, 2022-08-26:
So the reset is needed while we would like to keep the same approach as for other tests, where the device is already commissioned to the network.
For Cert Tests using TH, to run the bulk of tests, the commissioning is proceeding once for the tests session.
(no automated reset/factory reset between tests)

bzbarsky-apple, 2022-08-27:
Then after reset you would need to re-commission the device, no?  "WaitForCommissionee" just tried to establish CASE.  It does NOT do commissioning.  If you want to commission, you need to do one of the "Pair*" commands @doublemis1

bzbarsky-apple, 2023-10-16:
This is fixed now.

closed 2023-10-16 by bzbarsky-apple