project-chip/connectedhomeip #22038: Build for iOS failed

vtyutin, 2022-08-19:
#### Problem
I'm trying to build the sample app for iOS. It's failed with the following error. Please advice how to solve it. Mac OS version is 12.4

  error: subprocess-exited-with-error
  
  × Building wheel for gevent (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  Building wheel for gevent (pyproject.toml) ... error
  ERROR: Failed building wheel for gevent
  Building wheel for pyyaml (pyproject.toml) ... done
  Created wheel for pyyaml: filename=PyYAML-6.0-cp38-cp38-macosx_10_14_arm64.whl size=45338 sha256=e2ef3096204dce5c892285a8602bbf7cc6bd278daa716a6fe7cdd55056655a44
  Stored in directory: /Users/dns/Library/Caches/pip/wheels/52/84/66/50912fd7bf1639a31758e40bd4312602e104a8eca1e0da9645
  Building wheel for timeout_decorator (setup.py) ... done
  Created wheel for timeout_decorator: filename=timeout_decorator-0.5.0-py3-none-any.whl size=5008 sha256=e4d46416ed8114130932a39cd47bc738f9e8b609aba0d3e20edccbe18e94162b
  Stored in directory: /Users/dns/Library/Caches/pip/wheels/38/05/4e/161d1463ca145ec1023bd4e5e1f31cbf9239aa8f39a2a2b643
  Building wheel for docopt (setup.py) ... done
  Created wheel for docopt: filename=docopt-0.6.2-py2.py3-none-any.whl size=13707 sha256=3db9d64848e7b39a894c9121b8ac2c22deb802cd4d3e06702de62fb27dabe34a
  Stored in directory: /Users/dns/Library/Caches/pip/wheels/56/ea/58/ead137b087d9e326852a851351d1debf4ada529b6ac0ec4e8c
  Building wheel for greenlet (setup.py) ... done
  Created wheel for greenlet: filename=greenlet-1.1.0-cp38-cp38-macosx_10_14_arm64.whl size=105558 sha256=675e28cef0bff7d24b3f97e79fd8030d5a36ce60e2da603c4546834692a4f73b
  Stored in directory: /Users/dns/Library/Caches/pip/wheels/2f/e7/47/b6a893203ccff7cb1cb81576f22066a93657a6429a68a912d0
  Building wheel for ruamel.yaml.clib (setup.py) ... done
  Created wheel for ruamel.yaml.clib: filename=ruamel.yaml.clib-0.2.6-cp38-cp38-macosx_10_14_arm64.whl size=268974 sha256=f10f1eede737f9c25e34e327dc2c77f934cbae50a7eef6939a2a8f11a8f1e8a3
  Stored in directory: /Users/dns/Library/Caches/pip/wheels/17/0a/cb/8df6ed5547555605ad0f025f670be83ff322b3cfb5982e9427
Successfully built construct future reedsolo bitstring mobly cbor psutil watchdog stringcase tornado cffi pyyaml timeout_decorator docopt greenlet ruamel.yaml.clib
Failed to build gevent
ERROR: Could not build wheels for gevent, which is required to install pyproject.toml-based projects

 #### Proposed Solution
<suggested fix, suggested enhancement>

bzbarsky-apple, 2022-08-25:
> I'm trying to build the sample app for iOS.

@vtyutin using what exact steps?

mohitjadav86, 2022-08-29:
Also Facing Same Issue 

Building wheels for collected packages: gevent
  Building wheel for gevent (pyproject.toml): started
  error: subprocess-exited-with-error
  
  × Building wheel for gevent (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> See above for output.

woody-apple, 2022-08-31:
Can you please let us know what "the sample app" is as well as what steps you're using are? Please re-open with new details.

closed 2022-08-31 by woody-apple