project-chip/connectedhomeip #22040: IOS, Home Pod and rebooting devices

jonsmirl, 2022-08-19:
I have a Home Pod running the IOS beta software. It is attached to a Matter Device. Now I reboot the Matter device. After the Matter device reboots it just sits there displaying this...
```
E (726576) chip[IN]: Data received on an unknown session (LSID=41282). Dropping it!
E (729956) chip[IN]: Data received on an unknown session (LSID=41282). Dropping it!
E (732106) chip[IN]: Data received on an unknown session (LSID=41282). Dropping it!
E (735596) chip[IN]: Data received on an unknown session (LSID=41282). Dropping it!
```

I think what is going on here is that the Home Pod  can not tell the Matter device has been rebooted, so it does not know to establish a new session.   As long as the Matter device silently discards messages from this old session the Home Pod just assumes that it is dead.

Shouldn't the Matter device send a NAK of some kind back on that old session to give the Home Pod a clue to try again with a new session?

bzbarsky-apple, 2022-08-25:
Per spec, this is correct behavior.  There is no way to send a NAK back on an encrypted session if there are no encryption keys....

Eventually the homepod side will detect that it failed to deliver a message, mark the session as defunct, and create a new session the next time it needs one (though not all of this may be in the version you are using right now).

closed 2022-08-25 by bzbarsky-apple

jonsmirl, 2022-08-25:
If the user is tapping on the device in the Home app, the Home app should be aggressive about quickly detecting a reboot.

bzbarsky-apple, 2022-08-27:
The fastest that you can detect failure to deliver messages is when all the MRP sends fail, no?

jonsmirl, 2022-08-27:
The device could send an unencrypted advisory NAK packet back. Normally just discard this packet. But if the the Home app is in a retry loop you could look for it.  That keeps it from being used for DOS attack.

bzbarsky-apple, 2022-08-27:
> The device could send an unencrypted advisory NAK packet back

This was discussed during the spec discussions and rejected, fwiw, precisely because the DoS risks were too high.  In particular, since the packet can't be encrypted, anyone on the LAN can send it and prevent communication between devices.

jonsmirl, 2022-08-27:
It can only DOS you if you are looking for it. Don't look for except when you are retrying failed sends, in that case it is useful.

jonsmirl, 2022-08-27:
And it is not a general NAK, the source address of the packet has to match the destination address of the failed node you are retrying before you do anything with it.

jonsmirl, 2022-08-27:
Use case I am thinking about --- user frustrated with some part of the setup procedure who then keeps power cycling the device.

bzbarsky-apple, 2022-08-27:
> the source address of the packet has to match the destination address of the failed node you are retrying

Source addresses can be spoofed.

jonsmirl, 2022-08-27:
You're over thinking this. I can DOS the whole LAN just by flooding it with broadcast packets, not need to mess with a Matter device.

bzbarsky-apple, 2022-08-27:
I'm just telling you what the spec discussions looked like.  These are not my personal opinions.