project-chip/connectedhomeip #22062: [TC-I-2.3] Doesnt reproduce expected outcome in identify trigger-effect at step 2a .

Anshul272, 2022-08-22:
[TC_2.3_I_Fail.txt](https://github.com/project-chip/connectedhomeip/files/9392171/TC_2.3_I_Fail.txt)

**Steps to reproduce:**

Run Commission DUT to TH.

TH sends TriggerEffect command to DUT -> ./chip-tool identify trigger-effect 0 0 1 1

**Expected outcome :** 

If the DUT has to provision to verify the logs, Verify blink effect on the DUT all-cluster-app log 

1647942608.859355][4341:4341] CHIP:DMG:         ],
[1647942608.859392][4341:4341] CHIP:DMG:         
[1647942608.859420][4341:4341] CHIP:DMG:         InteractionModelRevision = 1
[1647942608.859448][4341:4341] CHIP:DMG: },
[1647942608.859523][4341:4341] CHIP:DMG: AccessControl: checking f=1 a=c s=0x000000000001B669 t= c=0x0000_0003 e=1 p=o
[1647942608.859574][4341:4341] CHIP:DMG: Received command for Endpoint=1 Cluster=0x0000_0003 Command=0x0000_0040
[1647942608.859615][4341:4341] CHIP:ZCL: RX identify:trigger effect 0x0 variant 0x0
[1647942608.859648][4341:4341] CHIP:ZCL: EMBER_ZCL_IDENTIFY_EFFECT_IDENTIFIER_BLINK

**Actual behavior:**

No blink effect is reflected

[1658985332.937423][3489:3494] CHIP:DMG: Received Command Response Status for Endpoint=1 Cluster=0x0000_0003 Command=0x0000_0040 Status=0x0
[1658985332.937462][3489:3494] CHIP:DMG: ICR moving to [AwaitingDe]
[1658985332.937525][3489:3494] CHIP:EM: Sending Standalone Ack for MessageCounter:236424845 on exchange 23382i

**[system configuration]([url]([url]([TC_2.3_I_Fail.txt](https://github.com/project-chip/connectedhomeip/files/9392159/TC_2.3_I_Fail.txt))))**
DUT: MG12+ RS9116(Lighting app)

Tested on: stash build: branch name: RC_august_12_2022

sowmyassp, 2022-08-25:
@Anshul272 You need to verify in DUT log "If the DUT has to provision to verify the logs, Verify blink effect on the DUT all-cluster-app log"

franck-apple, 2022-09-13:
v1.0 issue review: this looks old enough to not be a cert blocker. Closing. Please re-open if needed.

closed 2022-09-13 by franck-apple