project-chip/connectedhomeip #22074: Use of unitialized memory in GroupDataProviderImpl::SetKeySet

msandstedt, 2022-08-22:
#### Problem

As reported by @gharveymn.

This code:

https://github.com/project-chip/connectedhomeip/blob/b85c5ec60a1f9b35bcbe37b5b18c04e1d695db1e/src/credentials/GroupDataProviderImpl.cpp#L1589

From a caller like this:

https://github.com/project-chip/connectedhomeip/blob/b85c5ec60a1f9b35bcbe37b5b18c04e1d695db1e/src/app/clusters/operational-credentials-server/operational-credentials-server.cpp#L664

This means that keyset.epoch_keys[1] and keyset.epoch_keys[2] are used uninitialized.  Indeed, KeySet does not explicitly initialize these fields:

https://github.com/project-chip/connectedhomeip/blob/b85c5ec60a1f9b35bcbe37b5b18c04e1d695db1e/src/credentials/GroupDataProvider.h#L126
https://github.com/project-chip/connectedhomeip/blob/b85c5ec60a1f9b35bcbe37b5b18c04e1d695db1e/src/credentials/GroupDataProvider.h#L148

It's suspected that this will cause random data and length of data in serialized / stored keys.  This doesn't break anything, but isn't correct either.

 #### Proposed Solution

Make `EpochKey` initialize itself on construction, add the equivalent to `KeySet`, or make sure all call sites do this manually.

bzbarsky-apple, 2022-08-25:
The other option would be to not touch (not serialize, not deserialize, not assign as here) the fields outside `keys_count`/`num_keys_used`, right?

msandstedt, 2022-08-29:
> The other option would be to not touch (not serialize, not deserialize, not assign as here) the fields outside `keys_count`/`num_keys_used`, right?

True, but maybe even better would be to initialize the data and also not use it.

tcarmelveilleux, 2022-08-29:
This is a dupe of https://github.com/project-chip/connectedhomeip/issues/22054 with more details

tcarmelveilleux, 2022-08-29:
> The other option would be to not touch (not serialize, not deserialize, not assign as here) the fields outside keys_count/num_keys_used, right?

This could break existing commissioned devices which will have written 3 key slots, dependending how it's implemented. It has to be done carefully

msandstedt, 2022-08-29:
Dup of #22054 .

closed 2022-08-29 by msandstedt