project-chip/connectedhomeip #22086: IOS Commissioning Failed | mDNS discovery Timeout

LiwaaK, 2022-08-22:
# Background:
 - Commissioning ESP32 WRoom 32D through IOS 
 - USING: lighting-app example & CHIP Tool darwin
 - Commissioning Succeeds through CHIP TOOL on MacOS
 - Commissionin Succeeded a couple of times using CHIP TOOL on IOS (about 40 times). Commissioning then started failing 

# Problem:
- Commissioning finished & succeeded multiple times on IOS, then started failing
- Fail Error IOS:  `CHIPTool[1329:109531] Error retrieving device informations over Mdns: Error Domain=MTRErrorDomain Code=9 "Transaction timed out." UserInfo={NSLocalizedDescription=Transaction timed out., underlyingError=<MTRErrorHolder: 0x2817aafc0>}`
- Fail Error ESP32: `E (105380) chip[ZCL]: OpCreds: Got FailSafeTimerExpired`
 
# Exploring the persistance of the error:
- Changed Commisisoning Network Used, same error remained
- Tried other ESP32 examples, same error remained
- Tried using ESP32 Factory provider for the Certificate, same error remained
- Tried creating a new ESP32 matter app with customized clusters (was a simple on-off light) , same error remained
- Tried creating a new IOS matter app using Swift UI bridged from the Objective C matter framework using a bridging header, same error remained 

 
# IOS CHIP Tool commissioning log:
 `2022-08-22 23:06:07.715730+0300 CHIPTool[1329:109325] New SSID: KifahK.com Password: 03235059
2022-08-22 23:06:07.716165+0300 CHIPTool[1329:109325] [all] 🔵 [1661198767716] [1329:109325] CHIP: [CTL] Setting wifi credentials from parameters
2022-08-22 23:06:07.716203+0300 CHIPTool[1329:109325] [all] 🔵 [1661198767716] [1329:109325] CHIP: [CTL] Setting attestation nonce to random value
2022-08-22 23:06:07.716304+0300 CHIPTool[1329:109325] [all] 🔵 [1661198767716] [1329:109325] CHIP: [CTL] Setting CSR nonce to random value
2022-08-22 23:06:07.716396+0300 CHIPTool[1329:109325] [all] 🔵 [1661198767716] [1329:109325] CHIP: [CTL] Commission called for node ID 0x000000000000000B
2022-08-22 23:06:07.716420+0300 CHIPTool[1329:109325] [all] 🔵 [1661198767716] [1329:109325] CHIP: [CTL] Commissioning stage next step: 'SecurePairing' -> 'ReadCommissioningInfo'
2022-08-22 23:06:07.716438+0300 CHIPTool[1329:109325] [all] 🔵 [1661198767716] [1329:109325] CHIP: [CTL] Performing next commissioning step 'ReadCommissioningInfo'
2022-08-22 23:06:07.716453+0300 CHIPTool[1329:109325] [all] 🔵 [1661198767716] [1329:109325] CHIP: [CTL] Sending request for commissioning information
2022-08-22 23:06:07.716777+0300 CHIPTool[1329:109325] [all] 🟢 [1661198767716] [1329:109325] CHIP: [DMG] SendReadRequest ReadClient[0x120565220]: Sending Read Request
2022-08-22 23:06:07.717821+0300 CHIPTool[1329:109325] [all] 🔵 [1661198767717] [1329:109325] CHIP: [IN] Prepared secure message 0x16b1df680 to 0xFFFFFFFB00000000 (0)  of type 0x2 and protocolId (0, 1) on exchange 46752i with MessageCounter:5175733.
2022-08-22 23:06:07.717860+0300 CHIPTool[1329:109325] [all] 🔵 [1661198767717] [1329:109325] CHIP: [IN] Sending encrypted msg 0x16b1df680 with MessageCounter:5175733 to 0xFFFFFFFB00000000 (0) at monotonic time: 000000000148C742 msec
2022-08-22 23:06:07.717887+0300 CHIPTool[1329:109325] [all] 🟢 [1661198767717] [1329:109325] CHIP: [DMG] MoveToState ReadClient[0x120565220]: Moving to [AwaitingIn]
2022-08-22 23:06:07.978823+0300 CHIPTool[1329:109531] [all] 🔵 [1661198767978] [1329:109531] CHIP: [EM] Received message of type 0x5 with protocolId (0, 1) and MessageCounter:55635521 on exchange 46752i
2022-08-22 23:06:07.979204+0300 CHIPTool[1329:109531] [all] 🟢 [1661198767979] [1329:109531] CHIP: [EM] Found matching exchange: 46752i, Delegate: 0x120565230
2022-08-22 23:06:07.986869+0300 CHIPTool[1329:109531] [all] 🔵 [1661198767986] [1329:109531] CHIP: [CTL] ----- NetworkCommissioning Features: has WiFi. endpointid = 0
2022-08-22 23:06:07.988568+0300 CHIPTool[1329:109531] [all] 🔵 [1661198767988] [1329:109531] CHIP: [CTL] Successfully finished commissioning step 'ReadCommissioningInfo'
2022-08-22 23:06:07.988923+0300 CHIPTool[1329:109531] [all] 🔵 [1661198767988] [1329:109531] CHIP: [CTL] Commissioning stage next step: 'ReadCommissioningInfo' -> 'ArmFailSafe'
2022-08-22 23:06:07.989069+0300 CHIPTool[1329:109531] [all] 🔵 [1661198767989] [1329:109531] CHIP: [CTL] Performing next commissioning step 'ArmFailSafe'
2022-08-22 23:06:07.989301+0300 CHIPTool[1329:109531] [all] 🔵 [1661198767989] [1329:109531] CHIP: [CTL] Arming failsafe (60 seconds)
2022-08-22 23:06:07.992306+0300 CHIPTool[1329:109531] [all] 🟢 [1661198767992] [1329:109531] CHIP: [DMG] ICR moving to [AddingComm]
2022-08-22 23:06:07.992648+0300 CHIPTool[1329:109531] [all] 🟢 [1661198767992] [1329:109531] CHIP: [DMG] ICR moving to [AddedComma]
2022-08-22 23:06:07.993242+0300 CHIPTool[1329:109531] [all] 🔵 [1661198767993] [1329:109531] CHIP: [IN] Prepared secure message 0x16b523390 to 0xFFFFFFFB00000000 (0)  of type 0x8 and protocolId (0, 1) on exchange 46753i with MessageCounter:5175734.
2022-08-22 23:06:07.993456+0300 CHIPTool[1329:109531] [all] 🔵 [1661198767993] [1329:109531] CHIP: [IN] Sending encrypted msg 0x16b523390 with MessageCounter:5175734 to 0xFFFFFFFB00000000 (0) at monotonic time: 000000000148C855 msec
2022-08-22 23:06:07.994203+0300 CHIPTool[1329:109531] [all] 🟢 [1661198767994] [1329:109531] CHIP: [DMG] ICR moving to [CommandSen]
2022-08-22 23:06:08.100550+0300 CHIPTool[1329:109531] [all] 🔵 [1661198768100] [1329:109531] CHIP: [EM] Received message of type 0x9 with protocolId (0, 1) and MessageCounter:55635522 on exchange 46753i
2022-08-22 23:06:08.100964+0300 CHIPTool[1329:109531] [all] 🟢 [1661198768100] [1329:109531] CHIP: [EM] Found matching exchange: 46753i, Delegate: 0x121a08c68
2022-08-22 23:06:08.101207+0300 CHIPTool[1329:109531] [all] 🟢 [1661198768101] [1329:109531] CHIP: [DMG] ICR moving to [ResponseRe]
2022-08-22 23:06:08.101756+0300 CHIPTool[1329:109531] [all] 🔵 [1661198768101] [1329:109531] CHIP: [DMG] Received Command Response Data, Endpoint=0 Cluster=0x0000_0030 Command=0x0000_0001
2022-08-22 23:06:08.103537+0300 CHIPTool[1329:109531] [all] 🔵 [1661198768103] [1329:109531] CHIP: [CTL] Received ArmFailSafe response errorCode=0
2022-08-22 23:06:08.104008+0300 CHIPTool[1329:109531] [all] 🔵 [1661198768103] [1329:109531] CHIP: [CTL] Successfully finished commissioning step 'ArmFailSafe'
2022-08-22 23:06:08.104358+0300 CHIPTool[1329:109531] [all] 🔵 [1661198768104] [1329:109531] CHIP: [CTL] No NetworkScan enabled or WiFi/Thread endpoint not specified, skipping ScanNetworks
2022-08-22 23:06:08.104592+0300 CHIPTool[1329:109531] [all] 🔵 [1661198768104] [1329:109531] CHIP: [CTL] Commissioning stage next step: 'ArmFailSafe' -> 'ConfigRegulatory'
2022-08-22 23:06:08.104805+0300 CHIPTool[1329:109531] [all] 🔵 [1661198768104] [1329:109531] CHIP: [CTL] Performing next commissioning step 'ConfigRegulatory'
2022-08-22 23:06:08.104996+0300 CHIPTool[1329:109531] [all] 🔵 [1661198768104] [1329:109531] CHIP: [CTL] Setting Regulatory Config
2022-08-22 23:06:08.105442+0300 CHIPTool[1329:109531] [all] 🔵 [1661198768105] [1329:109531] CHIP: [CTL] Device does not support configurable regulatory location
2022-08-22 23:06:08.106768+0300 CHIPTool[1329:109531] [all] 🟢 [1661198768106] [1329:109531] CHIP: [DMG] ICR moving to [AddingComm]
2022-08-22 23:06:08.107086+0300 CHIPTool[1329:109531] [all] 🟢 [1661198768107] [1329:109531] CHIP: [DMG] ICR moving to [AddedComma]
2022-08-22 23:06:08.107931+0300 CHIPTool[1329:109531] [all] 🔵 [1661198768107] [1329:109531] CHIP: [IN] Prepared secure message 0x16b522b40 to 0xFFFFFFFB00000000 (0)  of type 0x8 and protocolId (0, 1) on exchange 46754i with MessageCounter:5175735.
2022-08-22 23:06:08.108264+0300 CHIPTool[1329:109531] [all] 🔵 [1661198768108] [1329:109531] CHIP: [IN] Sending encrypted msg 0x16b522b40 with MessageCounter:5175735 to 0xFFFFFFFB00000000 (0) at monotonic time: 000000000148C8C8 msec
2022-08-22 23:06:08.109422+0300 CHIPTool[1329:109531] [all] 🟢 [1661198768109] [1329:109531] CHIP: [DMG] ICR moving to [CommandSen]
2022-08-22 23:06:08.109796+0300 CHIPTool[1329:109531] [all] 🟢 [1661198768109] [1329:109531] CHIP: [DMG] ICR moving to [AwaitingDe]
2022-08-22 23:06:08.250527+0300 CHIPTool[1329:109676] [all] 🔵 [1661198768250] [1329:109676] CHIP: [EM] Received message of type 0x9 with protocolId (0, 1) and MessageCounter:55635523 on exchange 46754i
2022-08-22 23:06:08.251016+0300 CHIPTool[1329:109676] [all] 🟢 [1661198768251] [1329:109676] CHIP: [EM] Found matching exchange: 46754i, Delegate: 0x121a06408
[... cut ...]
E (397930) chip[DIS]: Failed to reply to query: 3000004
I (397940) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (397940) chip[DIS]: Failed to reply to query: 3000004
I (902430) chip[SVR]: Closing pairing window
I (902430) chip[DIS]: Updating services using commissioning mode 0
I (902430) chip[DIS]: CHIP minimal mDNS started advertising.
I (902440) NimBLE: GAP procedure initiated: stop advertising.

I (902450) chip[DL]: CHIPoBLE advertising stopped
I (902450) app-devicecallbacks: Current free heap: 103872

```

JourneyYoung, 2022-08-25:
I found darwin app sometimes  happend this issue when commissioning succeeded a couple of times. I guess when multi devices in the same network, this issue will happened.

LiwaaK, 2022-08-25:
@JourneyYoung I only had 2 devices commissioned on the fabric when this issue happened & they weren't even connected.  How does matter expect its protocol be adpoted as a standard if there are such basic reliability issues? It's supposed to release this fall. I think they most probably would delay it.

LiwaaK, 2022-08-28:
> @LiwaaK For this step:
> 
> > Commissionin Succeeded a couple of times using CHIP TOOL on IOS (about 40 times). Commissioning then started failing
> 
> what were the exact steps involved? Were >40 different devices used? Or was one device reset multiple times? Were any unpairing or reset steps performed on the iOS CHIPTool app?
> 
> It would be good to understand exactly how to reproduce this.

 40 is a rough number that represents the net number of times commissioning succeeded before the error happened. I only tested with 6 ESPs. I simply tested the process of commissioning about 40 times with different device certificates and fabric Ids. The number is meant to simply express that the error wasn't occuring initially.

During the first time this error appeared, about 5 devices were commissioned on the fabric. The 6th device failed. After a few tests with commissioning for a different Access point and still failing, I erased flash of off all ESPs, and created a new fabric.
 
With the new fabric having no devices commissioned, for the same AP, the commissioning failed. 

When trying with a new Access point, commissioning failed. There were 2 times were commissioning actually succeeded, during these:
- Disconnected the iOS phone from the Access point and reconnected it once mDNS broadcasting on the ESP32 started
- I couldn't recreate this more than 2 times though.

At this point,I had tried 2 different APs, and it failed, so I posted this issue.

I then tried to use one of the ESP32s as an Access point, and test commissioning devices to that AP. Commissioning Succeeded. 
- For 2 Access points that are home internet routers, commissioning failed
- For the ESP32 as an access point, commissioning succeeds. 

 It seems that it doesn't work with certain APs for certain reasons, I'm further exploring this.

noname202, 2022-12-26:
This issue seems to still be present. As described by @LiwaaK starts failing after successfully commissioning the ESP32 device many times before I get a `CHIP Error 0x000002022-12-26 20:44:54.450794+0100 CHIPTool[11618:1268044] Error retrieving device informations over Mdns: Error Domain=MTRErrorDomain Code=1 "Undefined error:11." UserInfo={underlyingError=<MTRErrorHolder: 0x2812fcd80>, NSLocalizedDescription=Undefined error:11., errorCode=11}`.

Tested with the All-Clusters-App example and the CHIP Tool App for IOS on commit [93da711](https://github.com/project-chip/connectedhomeip/search?q=93da7115e7c9790c4372938fe591e3f190d8c925&type=commits)

LiwaaK, 2022-12-27:
@noname202 The issue seems to be associated with certain access points that don’t reliably support mDNS. I haven’t delved deeper into this, maybe you should. 

Try different access points, using mobile data hotspot should work. Let me know the cases where it fails and succeeds.

noname202, 2022-12-27:
In my case it seems like the issue was that the Chip-Tool App was still somehow configured as a fabric on the device. Removing  it fixed the issue for me. However, unfortunately I didn’t have the time to investigate further, so please don‘t quote me on that.

bzbarsky-apple, 2023-01-03:
For what it's worth, "errorCode=11" means `CHIP_ERROR_NO_MEMORY`.

That's a strange situation to run into on iOS.  If you are able to share a sysdiagnose, that would be very helpful...

woody-apple, 2023-06-24:
Closing this given it looks like there's been no update in a while. If you're still hitting this please re-open with new logs!

closed 2023-06-24 by woody-apple