project-chip/connectedhomeip #22095: [SVe2 ] [TC-CC-1.1]featureMap value mismatch in lighting-app example

tulasi-silabs, 2022-08-23:
#### Problem
Getting the featureMap value mismatch in lighting-app example
Precondition:
Update the testHarness Version: v2.3 and Sha: ed69865  and Upload the latest pics file and  upload in to the testHarness
Steps:
1.Choose the test case TC-CC-1.1
2.Run the test case through testHarness

Expected:It should run all the test cases by maching the expected output as per the test case 
Actual: Found failure in the testHarness at step Read the global attribute:FeatureMap
DEBUG      | 2022-08-23 06:17:12.262812 |Found failure:  featureMap value mismatch: expected 0 but got 31
WARNING    | 2022-08-23 06:17:12.264695 | Test Failure:  featureMap value mismatch: expected 0 but got 31

Tested on:
Test Harness version:v2.3 and Sha: ed69865
Commit id : 7d424839ff11b8ddb9c656d0721730433c9d526a

Logs:
[UI_Test_Run_2022_8_23_11_46_56.log](https://github.com/project-chip/connectedhomeip/files/9400028/UI_Test_Run_2022_8_23_11_46_56.log)

sowmyassp, 2022-08-23:
@tulasi-silabs  We have use all-clusters-app for CC test-cases

cjandhyala, 2022-08-24:
Pls open issue in the scripts repo. this is not SDK issue, closing it here.

closed 2022-08-24 by cjandhyala

innovation-matters-iot, 2022-08-25:
https://github.com/CHIP-Specifications/chip-test-scripts/issues/406