project-chip/connectedhomeip #22107: TC-ACL-2.6 - AccessControlExtensionChanged event - ChangeType is not matching the expected outcome

sethunk, 2022-08-23:
In TC-ACL-2.6, 
In step-6, the following ACL entry is provided by changing one attribute in the subject field from the existing ACL entry

> ./chip-tool accesscontrol write acl '[{ "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null},
>  { "privilege": 3, "authMode": 3, "subjects": [111,222,333,555], "targets": [{"cluster":3 , "endpoint":2, "deviceType": null}]}]' 1 0

In step-7, when we read access-control-entry-changed event, As per spec it is expected to get the ChangeType as  Changed(0) or Removed(2) but we are getting the ChangeType as Added(1).

> Endpoint: 0 Cluster: 0x0000_001F Event 0x0000_0000
> [1661263187.514514][16437:16442] CHIP:TOO:   Event number: 9
> [1661263187.514517][16437:16442] CHIP:TOO:   Priority: Info
> [1661263187.514520][16437:16442] CHIP:TOO:   Timestamp: 16557989
> [1661263187.514526][16437:16442] CHIP:TOO:   AccessControlEntryChanged: {
> [1661263187.514530][16437:16442] CHIP:TOO:     AdminNodeID: 112233
> [1661263187.514533][16437:16442] CHIP:TOO:     AdminPasscodeID: null
> [1661263187.514537][16437:16442] CHIP:TOO:     **ChangeType: 1**
> [1661263187.514540][16437:16442] CHIP:TOO:     LatestValue: {
> [1661263187.514543][16437:16442] CHIP:TOO:       Privilege: 5
> [1661263187.514547][16437:16442] CHIP:TOO:       AuthMode: 2
> [1661263187.514550][16437:16442] CHIP:TOO:       Subjects: 1 entries
> [1661263187.514554][16437:16442] CHIP:TOO:         [1]: 112233
> [1661263187.514558][16437:16442] CHIP:TOO:       Targets: null
> [1661263187.514561][16437:16442] CHIP:TOO:       FabricIndex: 1
> [1661263187.514564][16437:16442] CHIP:TOO:      }
> [1661263187.514567][16437:16442] CHIP:TOO:     FabricIndex: 1
> [1661263187.514570][16437:16442] CHIP:TOO:    }
> [1661263187.514605][16437:16442] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_001F Event 0x0000_0000
> [1661263187.514608][16437:16442] CHIP:TOO:   Event number: 10
> [1661263187.514611][16437:16442] CHIP:TOO:   Priority: Info
> [1661263187.514614][16437:16442] CHIP:TOO:   Timestamp: 16557989
> [1661263187.514622][16437:16442] CHIP:TOO:   AccessControlEntryChanged: {
> [1661263187.514625][16437:16442] CHIP:TOO:     AdminNodeID: 112233
> [1661263187.514628][16437:16442] CHIP:TOO:     AdminPasscodeID: null
> [1661263187.514631][16437:16442] CHIP:TOO:     **ChangeType: 1**
> [1661263187.514635][16437:16442] CHIP:TOO:     LatestValue: {
> [1661263187.514638][16437:16442] CHIP:TOO:       Privilege: 3
> [1661263187.514641][16437:16442] CHIP:TOO:       AuthMode: 3
> [1661263187.514645][16437:16442] CHIP:TOO:       Subjects: 4 entries
> [1661263187.514649][16437:16442] CHIP:TOO:         [1]: 111
> [1661263187.514652][16437:16442] CHIP:TOO:         [2]: 222
> [1661263187.514656][16437:16442] CHIP:TOO:         [3]: 333
> [1661263187.514659][16437:16442] CHIP:TOO:         [4]: 555
> [1661263187.514663][16437:16442] CHIP:TOO:       Targets: 1 entries
> [1661263187.514669][16437:16442] CHIP:TOO:         [1]: {
> [1661263187.514672][16437:16442] CHIP:TOO:           Cluster: 3
> [1661263187.514675][16437:16442] CHIP:TOO:           Endpoint: 2
> [1661263187.514678][16437:16442] CHIP:TOO:           DeviceType: null
> [1661263187.514681][16437:16442] CHIP:TOO:          }
> [1661263187.514685][16437:16442] CHIP:TOO:       FabricIndex: 1
> [1661263187.514687][16437:16442] CHIP:TOO:      }
> [1661263187.514690][16437:16442] CHIP:TOO:     FabricIndex: 1

**Test Environment**

**App used** - allclusters app
**Platform** - Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM
**Network** - onnetwork
**Commit** -  b85c5ec60a1f9b35bcbe37b5b18c04e1d695db1e
**Spec Reference**  -  https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/ACL-Cluster.adoc#events

**Logs**
[TC_ACL_2.6.txt](https://github.com/project-chip/connectedhomeip/files/9403931/TC_ACL_2.6.txt)

mrjerryjohns, 2022-08-23:
The command that is being issued is not updating a given ACL entry - it is replacing it. The act of replacing an attribute's value for lists is represented as the combination of both a 'delete all' + 'add' operation in the SDK's implementation.

To actually cause a 'changed' event to get emitted, you'd need to update a _specific_ ACL entry. That isn't possible with chip-tool today.

mrjerryjohns, 2022-08-23:
Consequently, Step 7 is incorrect.

sethunk, 2022-08-24:
@mrjerryjohns based on your comment I feel that there is an issue in test plan of TC-ACL-2.6 step-7. So closing this issue.

closed 2022-08-24 by sethunk

sumaky, 2022-08-24:
@mrjerryjohns Could you let us know why the behavior is such that when we add or modify or remove  an ACL entry  all the exisiting ACL entries gets deleted and new entries and events gets generated?

@cjandhyala The specification doesn't mention any such behavior.