project-chip/connectedhomeip #22114: Fix incorrect initializing of CATs in matter python tests

mrjerryjohns, 2022-08-23:
#### Problem

With the exception of `TC_RR_1_1.py` that passes a specific value for `controller_cat_tags` to `default_matter_test_main`, the others don't.

This results in `MatterTestConfig.controller_cat_tags` having a default value of `None`. This gets passed into `FabricAdmin.NewController`, which requires the list be non-None in value, and have an empty list if there are no CAT tags specified.

That causes this execution assertion:

```
WARNING:FabricAdmin:Allocating new controller with CaIndex: 1, FabricId: 0x0000000000000001, NodeId: 0x000000000001B669, CatTags: None
Traceback (most recent call last):
  File "/Users/johnsj/work/devel/ext-git/connectedhomeip-2/src/python_testing/TC_SC_3_6.py", line 249, in <module>
    default_matter_test_main()
  File "/Users/johnsj/work/devel/ext-git/connectedhomeip-2/src/python_testing/matter_testing_support.py", line 716, in default_matter_test_main
    default_controller = stack.certificate_authorities[0].adminList[0].NewController(nodeId=matter_test_config.controller_node_id,
  File "/Users/johnsj/work/devel/ext-git/connectedhomeip-2/.environment/pigweed-venv/lib/python3.9/site-packages/chip/FabricAdmin.py", line 107, in NewController
    controller = ChipDeviceCtrl.ChipDeviceController(opCredsContext=self._certificateAuthority.GetOpCredsContext(), fabricId=self._fabricId, nodeId=nodeId,
  File "/Users/johnsj/work/devel/ext-git/connectedhomeip-2/.environment/pigweed-venv/lib/python3.9/site-packages/chip/ChipDeviceCtrl.py", line 172, in __init__
    c_catTags = (c_uint32 * len(catTags))()
TypeError: object of type 'NoneType' has no len()
Exception ignored in: <function ChipDeviceController.__del__ at 0x10cb8e040>
Traceback (most recent call last):
  File "/Users/johnsj/work/devel/ext-git/connectedhomeip-2/.environment/pigweed-venv/lib/python3.9/site-packages/chip/ChipDeviceCtrl.py", line 323, in __del__
    self.Shutdown()
  File "/Users/johnsj/work/devel/ext-git/connectedhomeip-2/.environment/pigweed-venv/lib/python3.9/site-packages/chip/ChipDeviceCtrl.py", line 287, in Shutdown
    if (self._isActive):
AttributeError: 'ChipDeviceController' object has no attribute '_isActive'
```

mrjerryjohns, 2022-08-23:
Justification for 1.0 inclusion:

- Platform change.
- Fixes certification tests.

andy31415, 2022-08-24:
V1 review: acceptable to have changes for this in master for v1, since it blocks/affects cert.

andy31415, 2022-09-01:
Seems fixed by #22116. Closing.

closed 2022-09-01 by andy31415