project-chip/connectedhomeip #22132: IM Reporting layer incorrectly applies hold-off for chunked messages

mrjerryjohns, 2022-08-24:
#### Problem

The minimum reporting interval is used to govern the minimum time separation between successive Report transactions. Within a given Report transaction, it's possible to send multiple ReportData messages in the case of chunked data.

The set of chunked messages that are part of the same logical ReportData action should not however be subject to minimum reporting intervals. However, our current implementation incorrectly applies that restriction.

To repro the problem:
1. Launch the REPL server
2. Setup a subscription with full wildcard to it from another chip-tool/repl instance with min-interval = 5s.
3. Create a new endpoint on the REPL server consisting of a a number of clusters, including the test cluster. This ensures it will exceed an MTU later.
4. Wait at least 5s from the time the sub was created.
5. Finalize the endpoint.

You should see chunked reports going out every 5s, when in fact, they should be sent immediately.

mrjerryjohns, 2022-08-24:
This will affect subscriptions to hubs mostly, especially when new endpoints come online.

mrjerryjohns, 2022-08-24:
We should add a unit-test to validate the eventual fix FYI.

andy31415, 2022-08-24:
V1 review: acceptable for master changes for v1 as this is fixing spec.

bzbarsky-apple, 2022-08-31:
Fixed by https://github.com/project-chip/connectedhomeip/pull/22139

closed 2022-08-31 by bzbarsky-apple