project-chip/connectedhomeip #22138: ASAN: found crash in ReadClient::Close

andy31415, 2022-08-24:
#### Problem
 
 ASAN found in https://github.com/project-chip/connectedhomeip/runs/7998105176?check_suite_focus=true
 
 ```
  READ of size 8 at 0x612000004290 thread T0
     #0 0x10868713a in chip::Messaging::ExchangeManager::GetSessionManager() const ExchangeMgr.h:184
     #1 0x108685918 in chip::app::ReadClient::CancelLivenessCheckTimer() ReadClient.cpp:817
     #2 0x1086855bb in chip::app::ReadClient::StopResubscription() ReadClient.cpp:71
     #3 0x108686781 in chip::app::ReadClient::Close(chip::ChipError, bool, bool) ReadClient.cpp:190
     #4 0x108685c5d in chip::app::ReadClient::~ReadClient() ReadClient.cpp:91
     #5 0x108686984 in chip::app::ReadClient::~ReadClient() ReadClient.cpp:90
     #6 0x1086869a8 in chip::app::ReadClient::~ReadClient() ReadClient.cpp:90
     #7 0x107506f0c in std::__2::default_delete<chip::app::ReadClient>::operator()(chip::app::ReadClient*) const unique_ptr.h:54
....
     #43 0x1074dbe25 in Commands::~Commands() Commands.h:28
     #44 0x1074d8f64 in Commands::~Commands() Commands.h:28
     #45 0x1074d8e60 in main main.cpp:49
     #46 0x7fff2043cf3c in start+0x0 (libdyld.dylib:x86_64+0x15f3c)
 
 0x612000004290 is located 80 bytes inside of 296-byte region [0x612000004240,0x612000004368)
 freed by thread T0 here:
     #0 0x10a79ea59 in wrap_free+0xa9 (libclang_rt.asan_osx_dynamic.dylib:x86_64+0x48a59)
     #1 0x10871ea16 in chip::Platform::MemoryFree(void*) CHIPMem-Malloc.cpp:111
     #2 0x108b468bf in void chip::Platform::Delete<chip::Messaging::ExchangeManager>(chip::Messaging::ExchangeManager*) CHIPMem.h:169
     #3 0x108b45ded in chip::Controller::DeviceControllerSystemState::Shutdown() CHIPDeviceControllerFactory.cpp:461
     #4 0x108ab7144 in chip::Controller::DeviceControllerSystemState::Release() CHIPDeviceControllerSystemState.h:160
     #5 0x108ab7009 in chip::Controller::DeviceController::Shutdown() CHIPDeviceController.cpp:349
     #6 0x108abbe4a in chip::Controller::DeviceCommissioner::Shutdown() CHIPDeviceController.cpp:495
     #7 0x108523b3a in CHIPCommand::ShutdownCommissioner(std::__2::basic_string<char, std::__2::char_traits<char>, std::__2::allocator<char> >) CHIPCommand.cpp:334
     #8 0x108523550 in CHIPCommand::MaybeTearDownStack() CHIPCommand.cpp:147
     #9 0x1085249b6 in CHIPCommand::Run() CHIPCommand.cpp:206
     #10 0x10857b2b0 in Commands::RunCommand(int, char**, bool) Commands.cpp:147
     #11 0x10857994d in Commands::Run(int, char**) Commands.cpp:51
     #12 0x1074d8e54 in main main.cpp:48
     #13 0x7fff2043cf3c in start+0x0 (libdyld.dylib:x86_64+0x15f3c)
 
 previously allocated by thread T0 here:
     #0 0x10a79e910 in wrap_malloc+0xa0 (libclang_rt.asan_osx_dynamic.dylib:x86_64+0x48910)
     #1 0x10871e8e0 in chip::Platform::MemoryAlloc(unsigned long) CHIPMem-Malloc.cpp:91
     #2 0x108b42e61 in chip::Messaging::ExchangeManager* chip::Platform::New<chip::Messaging::ExchangeManager>() CHIPMem.h:145
     #3 0x108b3f3a0 in chip::Controller::DeviceControllerFactory::InitSystemState(chip::Controller::FactoryInitParams) CHIPDeviceControllerFactory.cpp:166
     #4 0x108b3e3a6 in chip::Controller::DeviceControllerFactory::Init(chip::Controller::FactoryInitParams) CHIPDeviceControllerFactory.cpp:67
     #5 0x10851f923 in CHIPCommand::MaybeSetUpStack() CHIPCommand.cpp:104
     #6 0x1085244c9 in CHIPCommand::Run() CHIPCommand.cpp:189
     #7 0x10857b2b0 in Commands::RunCommand(int, char**, bool) Commands.cpp:147
     #8 0x10857994d in Commands::Run(int, char**) Commands.cpp:51
     #9 0x1074d8e54 in main main.cpp:48
     #10 0x7fff2043cf3c in start+0x0 (libdyld.dylib:x86_64+0x15f3c)
```

I assume this is somehow only related to failure paths, so not marking it as critical v1.

andy31415, 2022-08-24:
Maybe related to #19591, unsure (seems similar close and darwin platform)

bzbarsky-apple, 2022-08-25:
This is basically the same as https://github.com/project-chip/connectedhomeip/issues/20085

andy31415, 2022-08-25:
Closing as duplicate of #20085

closed 2022-08-25 by andy31415