project-chip/connectedhomeip #22144: Test Case [TC-CHANNEL-5.5] Change Channel By Number Verification (DUT as Client) throws IM Error 0x00000501: General error: 0x01 (FAILURE)

phonnakasturi-apple, 2022-08-25:
SHA 5471b35f09beafd7db4a5e6ccb9ca2e63d7c9010 from sve2_branch

Under [TC-CHANNEL-5.5] Change Channel By Number Verification (DUT as Client), when DUT sends ChangeChannelByNumber command to TH the following error is thrown

`throws [1661389739882] [10450:395004] CHIP: [TOO] Run command failure: IM Error 0x00000501: General error: 0x01 (FAILURE)`

**Command:** ./chip-tool channel change-channel-by-number 33 22 1 1

**Note:** This is observed in both chip-tool and darwin-framework-tool 

**Attaching the complete log:
From Accessory (raspi):**
[TC-CHANNEL-5.5-AccessoryLogs.txt](https://github.com/project-chip/connectedhomeip/files/9420412/TC-CHANNEL-5.5-AccessoryLogs.txt)

**From DUT (Mac):**
[TC-CHANNEL-5.5-DUTLogs.txt](https://github.com/project-chip/connectedhomeip/files/9420411/TC-CHANNEL-5.5-DUTLogs.txt)

krypton36, 2022-08-25:
This command is not supported on all-clusters-app. Please run on tv-app.

phonnakasturi-apple, 2022-08-25:
@krypton36  is run against chip-tv-app

bzbarsky-apple, 2022-08-25:
Channel `(33, 22)` is not supported by tv-app, as far as I can tell.  It supports channels `(9, 1)`, `(9, 2)`, and `(9, 3)`.  You can use `chip-tool channel read channel-list 1 1` to see the list of available channels.

Using `chip-tool channel change-channel-by-number 9 2 1 1` works.

The spec does not define the behavior when using "ChangeChannelByNumber" with a non-existent channel number.  FAILURE does not seem to be unreasonable.

@phonnakasturi-apple where did the `(33, 22)` channel number come from?

phonnakasturi-apple, 2022-08-25:
@bzbarsky-apple Channel numbers 33 , 22 were mentioned in the verification steps of SVE2_TestPlanVerificationSteps doc:

```
Product maker needs to provide instructions for how to trigger the command on the DUT. For comparison, the DUT behavior for this test step can be simulated using chip-tool (when DUT is a commissioner) or tv-casting-app (when DUT is a commissionee):

the
./chip-tv-casting-app channel change-channel-by-number 33 22 1 1

The log below shows the resulting log on the TH (tv-app or all-clusters-app):

[1658530848358] [21924:317544] CHIP: [DMG] InvokeRequestMessage =
[1658530848358] [21924:317544] CHIP: [DMG] {
[1658530848358] [21924:317544] CHIP: [DMG]         suppressResponse = false, 
[1658530848358] [21924:317544] CHIP: [DMG]         timedRequest = false, 
[1658530848358] [21924:317544] CHIP: [DMG]         InvokeRequests =
[1658530848358] [21924:317544] CHIP: [DMG]         [
[1658530848358] [21924:317544] CHIP: [DMG]                 CommandDataIB =
[1658530848358] [21924:317544] CHIP: [DMG]                 {
[1658530848358] [21924:317544] CHIP: [DMG]                         CommandPathIB =
[1658530848358] [21924:317544] CHIP: [DMG]                         {
[1658530848358] [21924:317544] CHIP: [DMG]                                 EndpointId = 0x1,
[1658530848358] [21924:317544] CHIP: [DMG]                                 ClusterId = 0x504,
[1658530848358] [21924:317544] CHIP: [DMG]                                 CommandId = 0x2,
[1658530848358] [21924:317544] CHIP: [DMG]                         },
[1658530848358] [21924:317544] CHIP: [DMG]                         
[1658530848358] [21924:317544] CHIP: [DMG]                         CommandFields = 
[1658530848358] [21924:317544] CHIP: [DMG]                         {
[1658530848358] [21924:317544] CHIP: [DMG]                                 0x0 = 33, 
[1658530848358] [21924:317544] CHIP: [DMG]                                 0x1 = 22, 
[1658530848358] [21924:317544] CHIP: [DMG]                         },
[1658530848358] [21924:317544] CHIP: [DMG]                 },
[1658530848358] [21924:317544] CHIP: [DMG]                 
[1658530848358] [21924:317544] CHIP: [DMG]         ],
[1658530848358] [21924:317544] CHIP: [DMG]         
[1658530848358] [21924:317544] CHIP: [DMG]         InteractionModelRevision = 1
[1658530848358] [21924:317544] CHIP: [DMG] },

```

bzbarsky-apple, 2022-08-25:
> Channel numbers 33 , 22 were mentioned in the verification steps of SVE2_TestPlanVerificationSteps doc:

Thanks.  Filed https://github.com/CHIP-Specifications/chip-test-plans/issues/2104

closed 2022-08-25 by bzbarsky-apple