project-chip/connectedhomeip #22149: [SVE2][TC-ACL-2.5] - It shows FabricIndex rather than AdminFabricIndex

AlvinHsiao, 2022-08-25:
In TC-ACL-2.5,
In step 5:  ./chip-tool  accesscontrol  read-event  access-control-extension-changed 1 0
The log shows FabricIndex rather than AdminFabricIndex in Draft_SVE2_TestPlanVerificationSteps.

[1661409734.979807][27141:27146] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_001F Event 0x0000_0001
[1661409734.979863][27141:27146] CHIP:TOO:   Event number: 3
[1661409734.979905][27141:27146] CHIP:TOO:   Priority: Info
[1661409734.979945][27141:27146] CHIP:TOO:   Timestamp: 4426028
[1661409734.980101][27141:27146] CHIP:TOO:   AccessControlExtensionChanged: {
[1661409734.980175][27141:27146] CHIP:TOO:     AdminNodeID: 112233
[1661409734.980227][27141:27146] CHIP:TOO:     AdminPasscodeID: null
[1661409734.980281][27141:27146] CHIP:TOO:     ChangeType: 1
[1661409734.980333][27141:27146] CHIP:TOO:     LatestValue: {
[1661409734.980388][27141:27146] CHIP:TOO:       Data: 1718
[1661409734.980431][27141:27146] CHIP:TOO:       FabricIndex: 1
[1661409734.980480][27141:27146] CHIP:TOO:      }
[1661409734.980534][27141:27146] CHIP:TOO:     **FabricIndex**: 1
[1661409734.980582][27141:27146] CHIP:TOO:    }

Test Environment

App used - lighting app
Platform - Chip-tool - RPI-4, 8GB RAM
DUT - CYW30739

Logs
[[TC-ACL-2.5]_CHIPTool_log.txt](https://github.com/project-chip/connectedhomeip/files/9422193/TC-ACL-2.5._CHIPTool_log.txt)

bzbarsky-apple, 2022-08-25:
Per spec, the event field here is called "FabricIndex".  Sounds like the verification steps are wrong?

@cjandhyala

bzbarsky-apple, 2022-08-25:
Filed https://github.com/CHIP-Specifications/chip-test-plans/issues/2102

cjandhyala, 2022-08-31:
@bzbarsky-apple I don't understand the issue , Spec talks about FabricIndex and VS also shows FabricIndex

bzbarsky-apple, 2022-08-31:
I don't know where @AlvinHsiao got AdminFabricIndex from, then....

AlvinHsiao, 2022-09-01:
@It's from Draft_SVE2_TestPlanVerificationSteps , Version 1.1

franck-apple, 2022-09-09:
closing, as the test plan issue was created.

closed 2022-09-09 by franck-apple