project-chip/connectedhomeip #22154: Error: Cannot find module 'cross-spawn'

ep1cman, 2022-08-25:
#### Problem
When trying to follow the following guide: https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/simulated_device_linux.md
I get the error below trying to run the first command

```
adlink@comhpcalt:~/connectedhomeip$  ./scripts/examples/gn_build_test_example.sh app1

  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
 ▀▀    █   █    ▀▀

  ACTIVATOR! This sets your shell environment variables.

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Python environment.....done
  Setting environment variables for pw packages............skipped
  Setting environment variables for Host tools.............done

Checking the environment:

20220825 13:23:05 INF Environment passes all checks!

Environment looks good, you are ready to go!

Searching for zcl file from /home/adlink/connectedhomeip/scripts/examples/../../examples/placeholder/linux/apps/app1/config.zap
node:internal/modules/cjs/loader:949
  throw err;
  ^

Error: Cannot find module 'cross-spawn'
Require stack:
- /home/adlink/connectedhomeip/third_party/zap/repo/src-script/script-util.js
- /home/adlink/connectedhomeip/third_party/zap/repo/src-script/zap-generate.js
    at Module._resolveFilename (node:internal/modules/cjs/loader:946:15)
    at Module._load (node:internal/modules/cjs/loader:787:27)
    at Module.require (node:internal/modules/cjs/loader:1012:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/home/adlink/connectedhomeip/third_party/zap/repo/src-script/script-util.js:17:19)
    at Module._compile (node:internal/modules/cjs/loader:1112:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1166:10)
    at Module.load (node:internal/modules/cjs/loader:988:32)
    at Module._load (node:internal/modules/cjs/loader:834:12)
    at Module.require (node:internal/modules/cjs/loader:1012:19) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/home/adlink/connectedhomeip/third_party/zap/repo/src-script/script-util.js',
    '/home/adlink/connectedhomeip/third_party/zap/repo/src-script/zap-generate.js'
  ]
}

Node.js v18.4.0
Traceback (most recent call last):
  File "/home/adlink/connectedhomeip/scripts/examples/../../scripts/tools/zap/generate.py", line 230, in <module>
    main()
  File "/home/adlink/connectedhomeip/scripts/examples/../../scripts/tools/zap/generate.py", line 218, in main
    runGeneration(zap_file, zcl_file, templates_file, output_dir)
  File "/home/adlink/connectedhomeip/scripts/examples/../../scripts/tools/zap/generate.py", line 141, in runGeneration
    subprocess.check_call(['node', './src-script/zap-generate.js', '-z',
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['node', './src-script/zap-generate.js', '-z', '/home/adlink/connectedhomeip/src/app/zap-templates/zcl/zcl.json', '-g', '/home/adlink/connectedhomeip/src/app/zap-templates/app-templates.json', '-i', '/home/adlink/connectedhomeip/scripts/examples/../../examples/placeholder/linux/apps/app1/config.zap', '-o', '/home/adlink/connectedhomeip/scripts/examples/../../zzz_generated/placeholder/app1/zap-generated']' returned non-zero exit status 1.
adlink@comhpcalt:~/connectedhomeip$ 
```

ep1cman, 2022-08-25:
I was able to resolve this by running:
```
cd third_party/zap/repo
npm ci
```

bzbarsky-apple, 2022-08-25:
Sounds like the documentation here needs to be updated to include "make sure you have a working ZAP"....  @vivien-apple?