project-chip/connectedhomeip #22155: About Apple Store Review

JourneyYoung, 2022-08-25:
#### Problem
I notice the darwin framework's name is "Matter.framework" and the newest iOS16 system contains this framework. Our App has contained matter.framework, when I tried to apply it to the TestFlight, I got this message:

"'Matter.framework' is not permitted. Do not include platform frameworks in your app."

We want thoes who user  iphones running on iOS15 or earlier  can use Matter, how can I make matter.framework in our app?

 #### Proposed Solution
Can I change the framework's name to use it?

JourneyYoung, 2022-08-27:
@bzbarsky-apple   Hello.  Do your team have plan to resolve this issue?

bzbarsky-apple, 2022-08-30:
@woody-apple Please take a look?

woody-apple, 2022-08-31:
Hi @JourneyYoung. Thanks for the issue! Matter.framework is shipping by default on the iOS platform (and has for a while), so feel free to use that if you can. That said, if you want to bring another framework named "Matter.framework" I encourage you to rename yours so there are no runtime conflicts with the system provided Matter.framework.

closed 2022-08-31 by woody-apple

CaoJieNan, 2025-06-27:
@JourneyYoung May I ask how you handled this issue? Did you directly rename the "Matter.framework"?

JourneyYoung, 2025-06-27:
@CaoJieNan just use Matter.framework which provide by Apple

CaoJieNan, 2025-06-27:
@JourneyYoung Understood, thank you for your response! If I still wish to use my self-compiled Matter.framework, do you have any recommendations?