project-chip/connectedhomeip #22173: Error handling between BindingManager::EstablishConnection and BindingManager::HandleDeviceConnectionFailure less then ideal

tehampson, 2022-08-25:
#### Problem
* Error handling in `BindingManager` seems like it is expecting different behaviors from `CASESessionManager`, or that CASESessionManager has evolved way beyond what BindingManager was initially expecting it to do. As a result the error handling between `BindingManager::EstablishConnection` and `BindingManager::HandleDeviceConnectionFailure` is no longer adequate. Seeing that 1.0 release is soon and we are only taking in changes that are more critical, this was not fixed and instead is documented here as a post 1.0 issue.

bzbarsky-apple, 2022-08-26:
BindingManager initially was created around a totally different setup in terms of session lifetimes than we have now... that is part of the problem for sure.

tehampson, 2022-09-19:
For any future reference to either myself or anyone else working on this bug, this is only one section that would need to be better thought out:
* https://github.com/project-chip/connectedhomeip/pull/22133#discussion_r956531756