project-chip/connectedhomeip #22174: Subscription Reliability

hnnajh, 2022-08-25:
The current design of Subscription is that subscription will be non-persisted by default and the subscriber should own "resubscribing" when an error/disconnect/reset happens.

From Spec:
```
When a subscriber discovers the loss of synchronization, it MAY then initiate a re-subscription
to resume the subscription.
```

and the way proposed to do that is using the subscription interval:
```
In all cases, the publisher eventually discovers the loss of synchronization by not receiving a Status
Response to a Report Data message that expects a response, or by receiving an error Status
Response.
```

However this method can leave some time gaps during which the subscribe might lose some of the events - even if the interval period is minimum. 

so i have some questions

- Does the SDK support persisting those subscriptions if the accessory manufacturer decides to
- Is there away to instruct the accessory to persist those subscriptions from the commissioner/admin side?
- If the admin/publisher reestablishes the subscriptions after losing the connection/reset/... , is there away to retrieve the "missed" events?
- Is there away from the admin/publisher side to detect a connection loss? instead of depending on subscription interval

bzbarsky-apple, 2022-08-26:
> Does the SDK support persisting those subscriptions if the accessory manufacturer decides to

Right now, no.

> Is there away to instruct the accessory to persist those subscriptions from the commissioner/admin side?

No.

> is there away to retrieve the "missed" events?

@hnnajh can you please define "missed"?

If the events are still in the publisher's event buffer, the subscriber will see them.  If they have been flushed from the buffer due to it filling up or due to the publisher rebooting, then they will be missed.

> Is there away from the admin/publisher side to detect a connection loss? instead of depending on subscription interval

Apart from non-delivery of some outgoing message or not receiving an expected incoming message, there's no way to do that, right?

bzbarsky-apple, 2023-03-13:
>   Does the SDK support persisting those subscriptions if the accessory manufacturer decides to

At this point the answer is yes.  See `CHIP_CONFIG_PERSIST_SUBSCRIPTIONS`.