project-chip/connectedhomeip #22177: 【Android】branch SVE-2 JNI error when commissioning device

TanZhibin, 2022-08-26:
branch SVE-2 chiptool.apk  find JNI error when commissioning device, then is crashed .
detail above see logs ,thanks

TanZhibin, 2022-08-26:
2022-08-26 11:05:38.327 21477-8498/com.google.chip.chiptool D/CTL: Successfully finished commissioning step 'ReadCommissioningInfo'
2022-08-26 11:05:38.327 21477-8498/com.google.chip.chiptool D/CTL: Commissioning stage next step: 'ReadCommissioningInfo' -> 'ArmFailSafe'
2022-08-26 11:05:38.327 21477-8498/com.google.chip.chiptool D/CTL: Performing next commissioning step 'ArmFailSafe'
2022-08-26 11:05:38.327 21477-8498/com.google.chip.chiptool D/CTL: Arming failsafe (60 seconds)
2022-08-26 11:05:38.327 21477-8498/com.google.chip.chiptool D/DMG: ICR moving to [AddingComm]
2022-08-26 11:05:38.327 21477-8498/com.google.chip.chiptool D/DMG: ICR moving to [AddedComma]
2022-08-26 11:05:38.327 21477-8498/com.google.chip.chiptool D/IN: Prepared secure message 0x75f2067010 to 0xFFFFFFFB00000000 (0)  of type 0x8 and protocolId (0, 1) on exchange 8004i with MessageCounter:221500907.
2022-08-26 11:05:38.327 21477-8498/com.google.chip.chiptool D/IN: Sending encrypted msg 0x75f2067010 with MessageCounter:221500907 to 0xFFFFFFFB00000000 (0) at monotonic time: 000000000071C962 msec
2022-08-26 11:05:38.327 21477-8498/com.google.chip.chiptool D/DL: Received SendWriteRequest
2022-08-26 11:05:38.328 21477-8498/com.google.chip.chiptool A/e.chip.chiptoo: java_vm_ext.cc:577] **JNI DETECTED ERROR IN APPLICATION: JNI NewByteArray called with pending exception java.lang.NoSuchMethodError: no non-static method "Lchip/devicecontroller/ChipDeviceController;.onCommissioningStatusUpdate(JLjava/lang/string;I)V"**
    java_vm_ext.cc:577]   at void chip.platform.AndroidChipPlatform.handleIndicationReceived(int, byte[], byte[], byte[]) (AndroidChipPlatform.java:-2)
    java_vm_ext.cc:577]   at void chip.platform.AndroidBleManager$1.onCharacteristicChanged(android.bluetooth.BluetoothGatt, android.bluetooth.BluetoothGattCharacteristic) (AndroidBleManager.java:126)
    java_vm_ext.cc:577]   at void com.google.chip.chiptool.bluetooth.BluetoothManager$getBluetoothGattCallback$1.onCharacteristicChanged(android.bluetooth.BluetoothGatt, android.bluetooth.BluetoothGattCharacteristic) (BluetoothManager.kt:156)
    java_vm_ext.cc:577]   at void android.bluetooth.BluetoothGatt$1$8.run() (BluetoothGatt.java:478)
    java_vm_ext.cc:577]   at void android.bluetooth.BluetoothGatt.runOrQueueCallback(java.lang.Runnable) (BluetoothGatt.java:780)
    java_vm_ext.cc:577]   at void android.bluetooth.BluetoothGatt.access$200(android.bluetooth.BluetoothGatt, java.lang.Runnable) (BluetoothGatt.java:41)
    java_vm_ext.cc:577]   at void android.bluetooth.BluetoothGatt$1.onNotify(java.lang.String, int, byte[]) (BluetoothGatt.java:472)
    java_vm_ext.cc:577]   at boolean android.bluetooth.IBluetoothGattCallback$Stub.onTransact(int, android.os.Parcel, android.os.Parcel, int) (IBluetoothGattCallback.java:306)
    java_vm_ext.cc:577]   at boolean android.os.Binder.execTransactInternal(int, long, long, int, int) (Binder.java:1170)
    java_vm_ext.cc:577]   at boolean android.os.Binder.execTransact(int, long, long, int) (Binder.java:1134)
    java_vm_ext.cc:577] 
    java_vm_ext.cc:577]     in call to NewByteArray
    java_vm_ext.cc:577]     from void chip.platform.AndroidChipPlatform.handleIndicationReceived(int, byte[], byte[], byte[])
2022-08-26 11:05:38.480 21477-8751/com.google.chip.chiptool E/DL: Select failed: 2000004
2022-08-26 11:05:38.547 21477-8498/com.google.chip.chiptool A/e.chip.chiptoo: runtime.cc:655] Runtime aborting...
    runtime.cc:655] Dumping all threads without mutator lock held
    runtime.cc:655] All threads:
    runtime.cc:655] DALVIK THREADS (33):
    runtime.cc:655] "Binder:21477_5" prio=5 tid=29 Runnable
    runtime.cc:655]   | group="" sCount=0 dsCount=0 flags=0 obj=0x13040208 self=0xb400007773a16890
    runtime.cc:655]   | sysTid=8498 nice=0 cgrp=default sched=0/0 handle=0x75f206dcc0
    runtime.cc:655]   | state=R schedstat=( 2757272141 478964907 4136 ) utm=128 stm=147 core=1 HZ=100
    runtime.cc:655]   | stack=0x75f1f76000-0x75f1f78000 stackSize=995KB
    runtime.cc:655]   | held mutexes= "abort lock" "mutator lock"(shared held)

Allen-Guof, 2022-09-01:
Same issue. Any update?

Allen-Guof, 2022-09-01:
@TanZhibin Hi, I try the codes and it works. 
https://github.com/project-chip/connectedhomeip/pull/22031

andy31415, 2022-09-06:
Closing as fixed by #22031

closed 2022-09-06 by andy31415