project-chip/connectedhomeip #22178: When the OTBR is rebooted after the Matter enddevice is paired with the OTBR, the chip tool command is not transmitted.

cw-20021349, 2022-08-26:
Hi

I am currently working on a Matter project.

I configured OTBR with Raspberry Pi, completed pairing with nRF52840 board (Light bulb), and confirmed that Light on/off works (use to chiptool command).

However, after rebooting the OTBR, the Chip tool light on/off command is not delivered to the Enddevice.

1. Successful commissioning of endevice and otbr for the first time (node id is 300)
./chip-tool-release onoff on 300 1 When sending command, node id 300 Fabric id 0x1A1EF49FC643541D

[1661490979.941199][1352:1352] CHIP:DIS: Verifying the received credentials
[1661490979.948676][1352:1352] CHIP:DIS: Added new fabric at index: 0x3, Initialized: 1
[1661490979.948805][1352:1352] CHIP:DIS: Assigned compressed fabric ID: 0x783E482064712CE1, node ID: 0x000000000001B669
[1661490979.972146][1352:1352] CHIP:DIS: Fabric (3) persisted to storage. Calling OnFabricPersistedToStorage
[1661490979.972271][1352:1352] CHIP:CTL: Joined the fabric at index 3. Compressed fabric ID is: 0x783E482064712CE1
[1661490980.130537][1352:1357] CHIP:TOO: Sending command to node 0x12c
[1661490980.131397][1352:1357] CHIP:DIS: Resolving 1A1EF49FC643541D:000000000000012C ...
[1661490980.137856][1352:1357] CHIP:DL: Avahi resolve found
[1661490980.138696][1352:1357] CHIP:DIS: Node ID resolved for 1A1EF49FC643541D:000000000000012C
[1661490980.140833][1352:1357] CHIP:DIS: UDP:[fd99:bf05:bbe9:1:1ed7:af3:6efd:a15c%wlan0]:5540: new best score: 6
[1661490980.140986][1352:1357] CHIP:DIS: Checking node lookup status after 10 ms
[1661490980.141612][1352:1357] CHIP:DIS: Keeping DNSSD lookup active
[1661490980.331406][1352:1357] CHIP:DIS: Checking node lookup status after 200 ms
[1661490980.332571][1352:1357] CHIP:SC: Initiating session on local FabricIndex 1 from 0x000000000001B669 -> 0x000000000000012C
[1661490980.335867][1352:1357] CHIP:IN: Prepared unauthenticated message 0x5585e74768 to 0x0000000000000000 (0) of type 0x30 and protocolId (0, 0) on exchange 4525i with MessageCounter:2314344470.
[1661490980.336651][1352:1357] CHIP:IN: Sending unauthenticated msg 0x5585e74768 with MessageCounter:2314344470 to 0x0000000000000000 at monotonic time: 00000000003D4264 msec
[1661490980.337620][1352:1357] CHIP:SC: Sent Sigma1 msg

2. Retransmit the following command after OTBT reboot
./chip-tool-release onoff on 300 1

Fabric id is different from fabric id before otbr reboot.

[1661491207.821921][789:789] CHIP:DIS: Added new fabric at index: 0x2, Initialized: 1
[1661491207.822056][789:789] CHIP:DIS: Assigned compressed fabric ID: 0x46C385D8286F4C18, node ID: 0x000000000001B669
[1661491207.845347][789:789] CHIP:DIS: Fabric (2) persisted to storage. Calling OnFabricPersistedToStorage
[1661491207.847646][789:789] CHIP:CTL: Joined the fabric at index 2. Compressed fabric ID is: 0x46C385D8286F4C18
[1661491207.848014][789:789] CHIP:CTL: Setting attestation nonce to random value
[1661491207.848207][789:789] CHIP:CTL: Setting CSR nonce to random value
[1661491207.852583][789:789] CHIP:CTL: Generating NOC
[1661491207.855649][789:789] CHIP:DIS: Verifying the received credentials
[1661491207.863075][789:789] CHIP:DIS: Added new fabric at index: 0x3, Initialized: 1
[1661491207.863209][789:789] CHIP:DIS: Assigned compressed fabric ID: 0x55F242389EDE75B3, node ID: 0x000000000001B669
[1661491207.874940][789:789] CHIP:DIS: Fabric (3) persisted to storage. Calling OnFabricPersistedToStorage
[1661491207.877534][789:789] CHIP:CTL: Joined the fabric at index 3. Compressed fabric ID is: 0x55F242389EDE75B3
[1661491208.155696][789:794] CHIP:TOO: Sending command to node 0x12c
[1661491208.155948][789:794] CHIP:DIS: Resolving D6D4D6EF92695544:000000000000012C ...
[1661491208.356344][789:794] CHIP:DIS: Checking node lookup status after 200 ms
[1661491213.160018][789:794] CHIP:DL: Re-trying resolve
[1661491218.156018][789:789] CHIP:DIS: Cancelling incomplete address resolution as device is being deleted.
[1661491218.156171][789:789] CHIP:DIS: Discovery does not require any more timeouts
[1661491218.158868][789:789] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.

I checked the topology of Otbr web and found it was still connecting.
[image]

When Otbr is rebooted, the n52840 board log is also output as shown below.

D: 4171091 [DL]OpenThread State Changed (Flags: 0x00000200)

I: 4171101 [DL]SRP Client was stopped, because current server is no longer detected.

D: 4171109 [DL]OpenThread State Changed (Flags: 0x00000200)

D: 4171116 [DL]OpenThread State Changed (Flags: 0x00000002)

D: 4171121 [DL]   Thread Unicast Addresses:

D: 4171125 [DL]        fd15:f73b:9669:466a:0:ff:fe00:0/64 valid rloc
[... 73 lines cut ...]
[1661491295.447558][800:805] CHIP:DIS: Resolving 1A1EF49FC643541D:000000000000012C ...
[1661491295.556267][800:805] CHIP:DL: Avahi resolve found
[1661491295.556467][800:805] CHIP:DIS: Node ID resolved for 1A1EF49FC643541D:000000000000012C
[1661491295.558047][800:805] CHIP:DIS: UDP:[fd99:bf05:bbe9:1:1ed7:af3:6efd:a15c%wlan0]:5540: new best score: 6
[1661491295.558151][800:805] CHIP:DIS: Checking node lookup status after 110 ms
[1661491295.558220][800:805] CHIP:DIS: Keeping DNSSD lookup active
[1661491295.648033][800:805] CHIP:DIS: Checking node lookup status after 200 ms
[1661491295.648186][800:805] CHIP:TOO: NodeId Resolution: 300 at UDP:[fd99:bf05:bbe9:1:1ed7:af3:6efd:a15c]:5540
[1661491295.648255][800:805] CHIP:TOO: MRP retry interval (idle): 5000ms
[1661491295.648320][800:805] CHIP:TOO: MRP retry interval (active): 300ms
[1661491295.648382][800:805] CHIP:TOO: Supports TCP: no
[1661491295.648600][800:805] CHIP:DIS: Discovery does not require any more timeouts
[1661491295.653085][800:800] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
[1661491295.653735][800:800] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-suUJsH)
[1661491295.656470][800:800] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1661491295.656650][800:800] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1661491295.656772][800:800] CHIP:DL: Inet Layer shutdown
[1661491295.656840][800:800] CHIP:DL: BLE shutdown
[1661491295.656920][800:800] CHIP:DL: System Layer shutdown

I think Enddevice and otbr can send and receive cluster commands successfully even if Otbr is rebooted if Fabric ID does not change after Otbr reboots

Please guide.

Thank you

bzbarsky-apple, 2022-08-26:
@cw-20021349 Is "OTBR" in this case the same device that chip-tool is running on?   Or a different device?

cw-20021349, 2022-08-29:
@bzbarsky-apple hi It's the same device.

cw-20021349, 2022-08-30:
Hi

When pairing with Chip-tool for the first time, config and pairing information are saved in /tmp.

When the Raspberry Pi reboots, the files in /tmp are deleted.

Chip-tool config files in /tmp after pairing with enddevice

[image]

After rebooting the raspberry, check the files in the /tmp path (all erased)
[image]

I think that the chip tool config and information when paired with Enddevcie for the first time should be saved in a path other than /tmp. Can you give me a guide?

bzbarsky-apple, 2022-08-30:
> @bzbarsky-apple hi It's the same device.

In that case the issue is that `chip-tool` stores its state in `/tmp`, as you discovered.  That's not particularly configurable right now, without code changes.

resghst, 2022-10-25:
I faced the same issue at Matter v1.0.0. 
I want to know how to fix or prevent this issue.

resghst, 2022-10-27:
I found out how to prevent this issue.

content:
A quick solution could be to modify the setting of /tmp:
sudo vim /usr/lib/tmpfiles.d/tmp.conf
Change the line "D /tmp 1777 root root -" to "d /tmp 1777 root root -"
Then the content of /tmp will not be removed on every reboot.

reference:
https://devzone.nordicsemi.com/f/nordic-q-a/91280/when-the-otbr-is-rebooted-after-the-matter-enddevice-is-paired-with-the-otbr-the-chip-tool-command-is-not-transmitted

closed 2023-05-08 by stale[bot]