project-chip/connectedhomeip #22193: Consider smarter timeout handling in PASESession

bzbarsky-apple, 2022-08-26:
#### Problem
Right now PASESession allows 30s for server-side processing (plus more time for MRP bits) for all its messages.

This might make sense for some of the messages that involve crypto, but for PBKDFParamRequest it seems like we could use a lower timeout; creating a PBKDFParamResponse is pretty cheap.  The only crypto operations there are hashing the incoming and outgoing messages....

This would allow faster detection of "advertisement is stale" situations.

 #### Proposed Solution
See above.

turon, 2022-08-29:
Agreed.  It seems very reasonable to call `mExchangeCtxt->SetResponseTimeout` on each message with a timeout for that message.  Perhaps we can bucket the messages into two or three categories of `low`, `medium`, and `high` complexity and have an appropriate timeout for each.

bzbarsky-apple, 2022-08-31:
I suspect at this point this will need to be post-1.0...

closed 2022-10-26 by woody-apple