project-chip/connectedhomeip #22195: Changes from PR 22104 break commissioning flows where the network info is provided later

bzbarsky-apple, 2022-08-26:
#### Problem
https://github.com/project-chip/connectedhomeip/pull/22104 changed things so we do only on-network discovery if we don't have network credentials at commissioning start.

But the infrastructure @chrisdecenzo put in place allows providing the network credentials later, async, after performing a scan for available networks, etc.  That setup is now broken as far as I can tell.

 #### Proposed Solution
Revert #22104.

@vivien-apple @woody-apple

vivien-apple, 2022-08-29:
I suspect this is related to the changes in #20766 ?

Can't we add a new field to `CommissioningParameters` such that we can rely on that to force the network scan if no credentials are provided ?
And in `chip-tool` an optional param can be added to enable/disable this flag...

bzbarsky-apple, 2022-08-29:
> I suspect this is related to the changes in https://github.com/project-chip/connectedhomeip/pull/20766 ?

Yes.

> Can't we add a new field to CommissioningParameters such that we can rely on that to force the network scan

Yes, we can.  That would also address this issue.  

The problem is not forcing a network scan, by the way.  That will happen if there are no credentials and we discovered over BLE.  The problem is not skipping BLE discovery.

andy31415, 2022-10-11:
@vivien-apple - is this fixed now given that #22300 is merged?

closed 2022-10-13 by bzbarsky-apple